opentx/opentx #2559: globale variable and exit channel min/max

nicolas81, 2015-07-24:
I set exit channel 1 max to VG3 and exit channel 2 min to -VG3 on my Taranis but when i import models in compagnion 2.1 the both exit channels min and max are set to VG1.

projectkk2glider, 2015-07-26:
@nicolas81 which version of OpenTX and Companion, which radio?

bsongis, 2015-07-26:
I think there is also a problem on export of Channels Min / Max when the users chooses VG3 / -VG3 (not only an export problem)

bsongis, 2015-07-26:
It is not a problem in mixes, but in channels limits, do I understand well the issue?

bsongis, 2015-07-26:
Next nightly builds should start in 4 hours, please give a test, thanks a lot!

closed 2015-07-26 by bsongis

nicolas81, 2015-07-27:
Sorry for the delay...
I uses companion 2.1 and the firmware opentx-taranis-2.1.0.
It's realy an export channels min/max and mixes seems ok on the radio.
(to test : with the radio, start a new model and set exit channels min/ max to VG2 or VG3 or VG4 etc. Import models into companion. You will see all the exit channels min/max set to VG1).
I am going to test the nightly build today and i tell you if the problem is fixed.
Thanks

nicolas81, 2015-07-27:
I have tested the nightly build of companion (without re-installing opentx on the radio).
Congratulation, the problem is fixed !
thanks for all

commit 71f8b663 (2015-07-26): Fixes #2559 - Needs test (next nightly build)