opentx/opentx #2560: Companion, Compare models, Curves not working

fraternl, 2015-07-25:
http://downloads-21.open-tx.org/nightly/companion/companionInstall_2.1.1N391.exe

When I do a Compare models of my Cumel, the curves are not shown.
Only Curve22 (which I don't use) is shown.

projectkk2glider, 2015-07-26:
Link to Cumel eepe https://cloud.githubusercontent.com/assets/8692856/8888344/0ab7bf10-32aa-11e5-9566-105b92f2ae1f.png

projectkk2glider, 2015-07-28:
This issue goes way back.

fraternl, 2015-08-20:
Seems to be fixed in https://github.com/opentx/opentx/issues/2734

LapinFou, 2015-08-20:
You wrote "partially fixed", what is not fixed yet?

fraternl, 2015-08-20:
Because this feature never worked I'm already glad I now have the raw data.
The curve graphs however are all placed on each other.
I assume they were meant to be apart, but not sure though....
In "print model" all the graphs overlap as well and a legenda is shown.
Maybe it should be turned into a feature request:
[image]

LapinFou, 2015-08-20:
OK. Thanks for the feedback.
I'm sure projectkk2glider will fix it one day. :)

fraternl, 2015-08-20:
He now just had a renewed acquintance with the code   ;-)

I made a new issue for this and this one can be closed
https://github.com/opentx/opentx/issues/2737

projectkk2glider, 2015-08-20:
Closing this one, the @bsongis is making an extensive refactoring of the code as a part of #2737.

closed 2015-08-20 by projectkk2glider