opentx/opentx #2562: Global function - Can't change track to play - Firmware 2.1.0

rguptaca, 2015-07-25:
Companion 2.1.1N390
Windows 8 (up to date)
Taranis (not the plus)
Vers: opentx-taranis-2.1.0
EEPR: 217

Global Functions:

Switch > Play Track > #voice.wav (the # is in the file name so that I can group sounds)

I select a new track but it keeps reverting back to the old track.

Please see video 
https://youtu.be/IfaWELSnpR4
<iframe width="560" height="315" src="https://www.youtube.com/embed/IfaWELSnpR4" frameborder="0" allowfullscreen></iframe>

kilrah, 2015-07-25:
What if you don't put a #? It's likely what causes the problem. Of course it's not normal, but you might be able to get around it by choosing a character that's part of the OpenTX set, ideally not a special one.

rguptaca, 2015-07-26:
I tried with the opentx wav files. Same.

kilrah, 2015-07-26:
That issue was fixed but you're using 2.1.0 firmware, not the nightly builds (you need to select the option for that in companion preferences).

closed 2015-07-26 by kilrah

rguptaca, 2015-08-05:
So that I don't waste people's time where can I view a list of all bugs, enhancements being worked on? I have a number of items I'd like to mention. I did a search of all firmware label and wondered if there was another way for me to check before I post.

bsongis, 2015-08-05:
For example this filter:
https://github.com/opentx/opentx/issues?utf8=%E2%9C%93&q=milestone%3A%22OpenTX+2.1.1%22+

rguptaca, 2015-08-05:
Thanks. Bookmarked. 

Rav

Have a wonderful day!

Ravi Gupta, PMP
Founder of Hike For Happiness
A Mental Health Fundraiser
http://hikeforhappiness.ca/
(519) 697-0796

https://github.com/opentx/opentx/issues?utf8=%E2%9C%93&q=milestone%3A%22OpenTX+2.1.1%22+