opentx/opentx #2564: Changes at languages file de.h.txt  from Next  #2488   are not transfered to Master

hrenz, 2015-07-26:
so at latest Master are some strings wrong and for 9XRPro to long

bsongis, 2015-07-26:
I changed my mind and currently only 'master' is maintained. We have enough work to make it bugfree, then we will start working on 2.2.0 (next)

closed 2015-07-26 by bsongis

kilrah, 2015-07-26:
E.g. do not do anything in next for now, what you want included should be dine in master. When we start working on next we'll base it on master's current state.

projectkk2glider, 2015-07-26:
>I changed my mind and currently only 'master' is maintained. We have enough work to make it bugfree, then we will start working on 2.2.0

Uff, I strongly agree with this decision. :+1:  The duplicate work was tiresome and error prone.

bsongis, 2015-07-26:
I forgot to answer to your post, but I followed your advice. In fact I didn't imagine how much work was remaining in 'master', I thought I would quickly dive into mixer rewrite in 'next'