opentx/opentx #2569: Google Earth export problem with renamed GPS values

projectkk2glider, 2015-07-26:
Related to #2546

The Google Earth export issue will keep reappearing, because users will inevitably rename some of the GPS data values. Then the Log Viewer will be unable to find them again. I have no idea how to solve this one.

griebd, 2015-07-26:
I don't user Google Earth so I may be talking nonsense here...

but isn't it as simple as making the user select the fields when he uses the export function in Log Viewer? at least on the first export...

flarssen, 2015-07-27:
Can't see why anyone would rename those. Just make them fixed by disabling the Name fields.

kilrah, 2015-07-27:
Find the n/e/s/w in the log and parse backwards from there? Nothing else would generate those.

flarssen, 2015-07-27:
The related issue was caused by a change in GPS altitude label.

kilrah, 2015-07-27:
For the altitude let the user select it, that way he can also use the baro alt if desired.

projectkk2glider, 2015-08-02:
In 70d67dd7dd84c8d8d6bb3dd4f83ce3ad3b2fbb3f the export was fixed to work with the new 2.1 telemetry names (GPS, GAlt and GSpd) and format. 

Next enhancement is to make this more flexible.

closed 2020-05-01 by kilrah