opentx/opentx #2570: Companion 2.1.1, Print model can't handle SF  "Reset Alt"

fraternl, 2015-07-26:
Companion 2.1.1, Print model can't handle SF  "Reset Alt"
Probably it can't handle more telemetry values.....

It was first mentioned in ticket https://github.com/opentx/opentx/issues/2541
I just closed that ticket because it involved another issue which was solved.

[image]
[image]
[image]

bsongis, 2015-07-26:
I will take this one, there is code missing and some refactoring to avoid duplicate ...

projectkk2glider, 2015-07-28:
This might also be fixed in #2590

bsongis, 2015-07-28:
@fraternl please reopen if needed after nightly builds

closed 2015-07-28 by bsongis

fraternl, 2015-07-29:
It's working fine....in 2.1.1N396