opentx/opentx #2572: [Taranis B][Companion sim] "Voice Language" = SE, Timer 1-3 says the wrong unit

483917, 2015-07-26:
Windows 8.1 pro Swedish
Companion 2.1.1N392 Eng/Swe
Taranis B:
opentx-taranis-haptic-timer3-lua-mixersmon-sqt5font-en-2.1.1N392.bin
opentx-taranis-haptic-timer3-lua-mixersmon-sqt5font-se-2.1.1N392.bin

First sorry for my bad english,,,

i have made my own SE-lang pack ,and it has the new numbering index from,
http://voices-21.open-tx.org/opentx-taranis/en/english-american-taranis.zip

when i test run the voice files on my laptop EN/SE and same index number all is correct.
And if i in "General Settings" Taranis and Companion, chose "Voice Language" = SE , Timer1 - 3, Countdown = voice, Minute Call = yes

The Timers1 - 3 "Minute Call" Says Seconds not Minutes
It is the same issue on , Taranis and Companion sim

I also tested renaming my /SOUNDS/se to /SOUNDS/en, but same issue the Timers says seconds not minutes.

here is my test model
https://www.dropbox.com/s/4rsxgkjaqb50525/MODEL18-2015-07-26.bin?dl=0

kilrah, 2015-07-26:
Hmm I don't think the SE TTS system was ever finished or checked... Each language has its own handling file and code/numberings may differ depending on the language's "logic".

483917, 2015-07-26:
Ok thanks,, I did a quick dirty fix for the moment, I Selected Voice Language = English and renamed my /SOUNDS/se to /SOUNDS/en on Taranis SD-card, now timers1-3 says minutes not seconds and counting down from 30,20,10,9,8,,,, in swedish . Maybe some other sounds are wrong now ??

closed 2020-07-08 by stale[bot]