opentx/opentx #2576: Companion print model and compare model has missing receiver number

fraternl, 2015-07-27:
When I was comparing 2 models I noticed that it didn't flag the receiver numbers.
In fact, these are completely missing.

Not really a bug, but if it's used for some purposes it could be valuable info.
I wouldn't mind that much if it's not added, but for completeness I think it should (low priority?)

kilrah, 2015-07-27:
There's a n open issue for pretty much a complete rewrite of compare and print dialogs because they're quite outdated and code is duplicated. Not worth fixing much until then I'd say.
https://github.com/opentx/opentx/issues/1063

fraternl, 2015-07-27:
Ah, ok....
As long as the "print model" and "compare model" isn't just scratched before a new one is put in place. I find it quite valuable...

projectkk2glider, 2015-08-02:
Related to #1063

commit 6ac81107 (2015-08-23): Fixes #2576: display receiver number in model print/compare
closed 2015-08-23 by projectkk2glider