opentx/opentx #2579: Checking trainer input validity

raphaelcoeffic, 2015-07-27:
Gentlemen,

As of now, I've seen that the validity of the trainer input signal is tested properly in the code (ppmInValid is set in such case), and the trainer mode is deactivated properly, thus returning the control to the instructor. However, the issue tracking this feature (#36) was referring to the possibility to trigger some sound in such a case.

The only thing I've seen so far to implement this feature was to define a couple of logical switches checking for TR1 to TR2 being 0. Apart from the fact that this event is not really singular (could happen by accident, just leave the sticks in the middle ;-)), it is probably not really reliable.

So now my question: is there any way to directly check in a valid trainer input is available or not? I have no idea what the best way to implement that would be, btw. what the correct and logical way would be. Maybe a logical switch condition? I could implement it myself if you guys just point me into the right direction.

raphaelcoeffic, 2015-07-27:
Or maybe as an option to the "Trainer" special/global function? I tried those global functions a couple of minutes ago and am already addicted to it ;-)

kilrah, 2015-07-27:
Yeah we could have a "trainer lost/recovered" like for telemetry

By the way the validity check is apparently lacking on the alternate inputs (module/battery bay, SBUS module for sure), that should be added too.

raphaelcoeffic, 2015-07-27:
Ok, then I'll start with that part, as I'll need the SBUS input as well. Should I add a new issue for that bit?

raphaelcoeffic, 2015-07-28:
Hey! From what I could find, it seems that the PPM inputs are checked exactly the same way (exact same code, which might be factorised, by the way ;-)):
- in opentx.cpp: ISR(TIMER3_CAPT_vect)
- in targets/sky9x/board_sky9x.cpp: TC3_IRQHandler()
- in targets/taranis/trainer_driver.cpp: TIM3_IRQHandler()

Concerning the SBUS input, it is handled the same way everywhere, which means, pushed in a FIFO, and then polled/decoded periodically from the main loop. Validating the SBUS input in general can only be restricted to the framing & decoding, as the full 11 bits are meaningful. By the way, it might be interesting to decode the additional bits (frame lost & failsafe), which could be of interest for the wireless trainer use-case, which is why the SBUS has been included in the first place, right? With my X8R & X4R receivers, the more frames marked with the "frame lost" bit, the worse is the signal. When the signal is gone, all the frames are marked with the "failsafe" bit. In this case, we might want to disable the trainer input, as we are getting receiver's failsafe values anyway.

So here is what I would suggest:
 - trigger an event on (ppmInValid == 0)|(ppmInValid != 0) in the same fashion as "telemetry lost"/"telemetry recovered".
 - ignore "failsafe" SBUS frames, thus letting ppmInValid go down to zero and finally invalid the input.
 - compute the percentage of "frame lost" marked frames, and feed that value to the telemetry inputs.

By the way, is there any place other than here where you guys discuss opentx development, so that I could get some bootstrap? 

Cheers,
Raphael.

bsongis, 2015-07-28:
Right  "trainer lost/recovered" would be nice!

PR #2606 (2015-08-01): Fixes #2579: Added support for Trainer signal alarm
- added alarm “Trainer signal lost” & “Trainer signal recovered”.
- moved PPM capture code to trainer_input.{h,cpp}

This is the second try (see earlier push request #2593).
(Fixes #2579)
commit db072442 (2015-08-01): Merge pull request #2606 from opentx/raphaelcoeffic/issue2579

Fixes #2579: Added support for Trainer signal alarm
commit f6bff968 (2015-08-01): Fixes #2579: Added support for Trainer signal alarm

- added alarm “Trainer signal lost” & “Trainer signal recovered”.
- moved PPM capture code to trainer_input.{h,cpp}
closed 2015-08-01 by bsongis