opentx/opentx #2580: Clicking "Show Details" whilst syncing SD causes Companion to hang

fraternl, 2015-07-27:
Clicking
[image]
 "Show Details" whilst syncing SD causes Companion to hang

fraternl, 2015-07-28:
I used "Synchronize SD" today and now on my setup the "Show Details" is checked by default.
I will get the same "not responding", but after a while it comes back and everything is done.

I would make this low priority unless you know of an easy fix.

rguptaca, 2015-08-07:
I've been using 2.1.1N bills and now 2.1.1 released to the public. On both Windows & and OS X synchronize does not work. On windows it appears to hang. On OS X it shows complete within seconds but nothing actually happens for the source or destination folders.
For Windows I am using an xcopy script for now.
Actually during my testing I removed the SD card form my Taranis and busted it. Maybe I should have disconnected the battery first. Lucky I had made a copy backup and all is good now. 8GB in a Tx is ridiculous:)

ReneTr, 2015-08-10:
Just to add some information: I used the sync SD today and it also got the Not responding message in Windows 10. However after a couple of seconds it showed what it was doing in really tiny print. (See 1st screenshot)
After a while it completed successfully (I checked and the files did sync) and the final screen was normal. (2nd screenshot)

[image]
[image]

bsongis, 2015-08-10:
Yes, this is only a threading issue, there are many possibilities, either have a dedicated thread (rsync thread different from UI thread), or make the rsync asynchronous and add its calls inside a Qt event loop. I will take care of it.

commit f5f1a052 (2015-08-17): Fixes #2580
closed 2015-08-17 by bsongis