opentx/opentx #2584: 'Speaker' is not compatible with 2.1

JoeB99, 2015-07-28:
http://www.open-tx.org/downloads.html

Speaker offered here will not generate the sounds required for 2.1. Speaker should be open-sourced, and/or updated to work with 2.1.

jtaylor2, 2015-07-28:
According to https://github.com/opentx/opentx/blob/master/sound/speaker/Properties/AssemblyInfo.cs it is Open Source so feel free to update it for 2.1.

/* This file is part of OpenTX Speaker.
* OpenTX Speaker is free software: you can redistribute it and/or modify
* it under the terms of the GNU General Public License as published by
* the Free Software Foundation, either version 3 of the License, or
* (at your option) any later version.

JoeB99, 2015-07-28:
Perfect thanks, i dont know why i couldn't see the speaker code!

i'll have a crack at changing the sentence tables, here:
https://github.com/opentx/opentx/blob/master/sound/speaker/sentencetables.cs

jtaylor2, 2015-07-28:
After you opened the incident I was looking at it and thinking of doing that, but I don't have Visual Studio installed so was trying to build  it from the csproj file  using the .net\framework MSBuild.exe command, but am getting errors.  I don't know c# anyway so just as well you do it.

JoeB99, 2015-07-28:
I dont know CS either :'( I was thinking of just modifying the other_sounds.txt and system_sounds.txt and removing the sentence table code as its not required, the software looks for a folder structure with these two text files and it reads them in to the program.
This way, to add a language or change filenames, text etc all that is needed is a modification to the text files. That is way easier to maintain than software code as anyone can do it.... What do you think?

bsongis, 2015-07-28:
This is a big piece of duplicate code, all those sounds are already defined somewhere else!

JoeB99, 2015-07-28:
Great. Where?

Spainman, 2016-04-24:
Are the tts_xx.py  in /radio/util the reference? Will the csv files stay available online with the sound packs as it seems to be now?
Speaker could operate on csv files (load + save) instead of using the current hard coded mess. I would have a look at it if the csv download will stay available and up to date for the foreseeable future...

kilrah, 2016-04-24:
tts_xx.cpp in radio\src\translations are the references for what actually happens in the radio.
tts.py is what generates all the packs and csv's, which hopefully matches the code but that currently has to be taken care of manually.

Spainman, 2016-04-24:
The source and header files you pointed to don't seem to help too much when it comes to automating the sound file generation....

So to sum it up: Right now the csv files reflect the contents of the manually maintained, therefore possibly not always up-to-date, but still 'officially current' sound packs, right? If it is good enough for the sound packs people are pointed to by companion then it should be good enough for the speaker app or am I missing something?

kilrah, 2016-04-24:
Yep it should be correct.

JoeB99, 2016-04-26:
How / what was used to create the sounds when 2.1 was released? Surely someone would have had to make the sounds originally.?

kilrah, 2016-04-27:
tts.py.

JoeB99, 2016-12-20:
Shameless bump on this as i'm still using 2.0 because i cannot create custom sounds for 2.1 as speaker was not updated with the correct file list.
The decision to reorder the voice files from 2.0 -> 2.1 is a real pain as it breaks speaker.

Has anyone managed to use a voice generator that contains a list of 2.1 voice files?

kilrah, 2016-12-20:
This is the go-to voice generator nowadays:
https://github.com/CaffeineAU/TTSAutomate

some people have posted sound lists for 2.1 and 2.2 on rcgroups, it's also easy to make it yourself from the list supplied in the official 2.1 SD card image.

JoeB99, 2016-12-20:
Yes that looks like it would work, i cant find a SD card image on the downloads page http://www.open-tx.org/downloads, I have downloaded a voices zip, and there is no text file list in there.

closed 2019-08-19 by kilrah