opentx/opentx #2591: Allow mixes to be switched "OFF".

Julez1, 2015-07-28:
Hello,

what I am missing currently is the clean and simple option to switch a mixer off. Do do this now, I would have to delete the mixer, set the weight to 0, create a logic switch which is always "off" and assign the mixer to it, or tie the mixer to a flight phase that is never used.
A clean and simple solution would be either to have a dedicated checkbox switching the mixer off, or being able to choose "OFF" in the switch menu.
Apparently someone had the idea before me, because in er9x it was possible to switch "OFF".
Regards,
Julian

[image]
[image]

borland1, 2015-07-28:
Won't someone confuse that switch option?  The power switch is labeled "ON".

Julez1, 2015-07-28:
I'd be very happy either way. But another advantage of the check box would be that one would not have to reassign a previously preferred switch when reactivating that mixer.

fraternl, 2015-07-28:
If this is implemented (the checkbox version, I hope) it should be made clear somehow that this mixer-line is turned off in the display...

kilrah, 2015-07-29:
We can have ot greyed out, I think we even already have code for that in the UI.

borland1, 2015-08-02:
Greyed out text or strikethrough text would work.
Most Special Functions have a enable check box. A disable check box on the Main Mixer page might allow for easier way to disable/enable.
I wonder how the Multiplex option might work for mixer lines below one set to 'disabled'?  I guess the intent would be like "Multiplex: Bypass".

fraternl, 2015-08-02:
Either they should be grayed out too as some confirmation and skipped or just left to the responsibility of the user and stay enabled.

It took a long time for someone (julez) to come up with this request. I doubt that many will make use of it. I'd think it should be left to the user. OpenTX is not a system that holds your hand all the time.

The example he gave would have been better off with a logical switch.

kilrah, 2015-08-02:
From what I can see there is no free bit in the mixer structure that could be reused for a checkbox so that couldn't be done before 2.2, and even then I doubt it's worth "opening" a whole new byte just for it.

Switch set to "OFF" is possible, but then we'd have 2 ONs (nothing and ON). We could remove ON, but then we'd lose quick access to OFF from the shortcut popup.

Multiplex set to OFF should be possible and the easiest/cleanest.

closed 2019-11-15 by stale[bot]