opentx/opentx #2595: Add a CLI

bsongis, 2015-07-28:
On all ARM boards which have a serial port.
Enabled with CLI=YES in the Makefile

raphaelcoeffic, 2015-07-29:
Sounds like a good idea, but it depends on the info/commands available with such an interface. Talking about which, I'd love to have the USB interface offer a CDC device on runtime (maybe only when CLI and/or DEBUG is enabled). That would enable anyone to rapidly debug things with just the USB cable used to flash the TX otherwise. It seems that ST has some libs for that.

kilrah, 2015-07-29:
Yes I'd rather see it over USB too!

projectkk2glider, 2015-07-29:
We need USB serial port anyway to implement other protocols, like model transfer, firmware update, telemetry transfer, etc. There is even an issue for this somewhere.

bsongis, 2015-07-29:
I do agree. The massstorage protocol is not the right way to flash firmware and EEPROM. We could also have massstorage enabled when the Taranis is ON, showing many USB devices (joystick / serial / massstorage). And keep the bootloader minimal.

projectkk2glider, 2015-07-29:
>  We could also have massstorage enabled when the Taranis is ON, showing many USB devices (joystick / serial / massstorage). 

We might have a problem there. The stm32f2x has only 1 control and 3 bi-directional USB endpoints. The HID interface (joystick) needs 1, the serial port needs two. So right there we used them all up.

One possible solution is to have a radio wide setting that decides which USB interface gets initialized when USB cable is connected. So we would emulate only one USB type device at the time. This would also immensely simplify the development of USB code. I have looked at a few examples of the so called composite USB device (that is the device that simultaneously implement two or more USB devices) and it is not pretty.

bsongis, 2015-07-29:
Or we raise a popup when the cable is plugged in to ask the user what he wants to do. But I think I will prefer your solution :)

griebd, 2015-07-29:
more work to do, but a mix of both would be better in my opinion.
by default you have a popup for user selection. And for those who always choose the same option, a setting to bypass the popup and initialize the chosen USB interface.

bsongis, 2015-07-29:
yes more work, but  perhaps not a bad idea

projectkk2glider, 2015-07-29:
The decision at the point of USB cable insertion is I am afraid too late. 

It think one of the problems we sometimes have with the current USB connection (some slowness in the beginning) is because we initialize the USB infrastructure only after we detect the USB cable connection. This might already be too late. Because of that radio can miss the first opportunity to talk to the USB host device. 

I am now testing if having USB infrastructure initialized from the radio start will improve this situation. If that proves to be the case, then we have no choice other than to implement USB device type as a radio setting only.

kilrah, 2015-07-29:
Why old-school serial over usb and all its problems when we could do everything through the driverless HID?

raphaelcoeffic, 2015-07-29:
But that would require to use a specific software rather than the plain normal combination of drivers/software delivered by the OS. Doesn't it?

projectkk2glider, 2015-07-29:
>Why old-school serial over usb and all its problems when we could do everything through the driverless HID?

Please elaborate, I am :ear:

kilrah, 2015-07-29:
I don't know all the implementation details but I know several devices I use are detected as generic HID when connected to the computer, don't need any driver other than the OS' built-in HID class drivers, and their companion apps can happily exchange data both ways with the device without requiring any install.

Compare that to installing device-specific and OS-specific virtual COM port drivers, the user having to find and select said correct COM port number etc...

bsongis, 2015-07-29:
Perhaps he means CDC ACM as in PX4 Bootloader:
https://github.com/PX4/Bootloader

raphaelcoeffic, 2015-07-30:
http://www.libusb.org/wiki/FAQ#CanIcreateadriverlessdeviceusingHIDclass

raphaelcoeffic, 2015-07-30:
According to this source, we should rather go with a vendor specific device class to be able to implement all the nice features in a companion app. The device itself is then accessed directly by libusb, which is already used by numerous devices and apps.

However, I don't think this was the purpose of a CLI. For me, the upside of a CLI is that you can access from any terminal program on any OS, script it, etc... And, I don't have to use a GUI program all the time, which makes me way more productive when testing/debugging with very repetitive processes. Typically: compile, upload, start a feature, and see how it works / watch the debug output. If I can do everything using the keyboard, I'm way faster than jumping between keyboard and trackpad, typing, pointing and clicking. But that might be just my old-school view on the topic. I understand the appeal for most users to just use a nice GUI program without installing tons of drivers, looking for virtual COM port numbers, etc... Especially on windows, this can become a real pain-in-the-ass (which is one the reasons I don't use it anymore ;-)).

projectkk2glider, 2015-07-30:
For our implementation of USB serial port, we can use ST's STM32 Virtual COM Port Driver (for Windows) http://www.st.com/web/en/catalog/tools/PF257938#

raphaelcoeffic, 2015-07-30:
An other OS will use the native CDC driver? Linux & OS X have standard CDC/ACM drivers for USB modems.

raphaelcoeffic, 2015-07-30:
@projectkk2glider: if you have anything working right now, just push a branch and I'll test it on OS X. I believe it should work out-of-the-box.

projectkk2glider, 2015-07-30:
> An other OS will use the native CDC driver?

Yes, at least on Linux, where I test/develop. 

>if you have anything working right now, just push a branch

Can't do, files are at another location.

raphaelcoeffic, 2015-07-30:
No problem! Just wanted to let you know that I could help with OS X, in case needed ;-)

projectkk2glider, 2015-07-30:
@raphaelcoeffic the first example is up, use branch https://github.com/opentx/opentx/tree/projectkk2glider/usb_serial 

You must build with the option `USB=SERIAL`. When you connect with the terminal to radio's USB serial port and type a character, you will get back that same character and a `-` character. For example: you type `abcd`, you get back `a-b-c-d-`.

I have also tested the ST driver for USB serial port and it works! Driver is here http://www.st.com/web/en/catalog/tools/PF257938 For WinXP I used Win7 driver package from that bundle.

raphaelcoeffic, 2015-07-30:
Any baud-rate that should be set for the terminal?

bsongis, 2015-07-30:
It depends, on UART3 it could be 9600bauds to allow a bluetooth connection

bsongis, 2015-07-30:
But ideally it should be configurable of course.

raphaelcoeffic, 2015-07-30:
Works like a charm!

FrSky Taranis Serial Port:

  Product ID:	0x5740
  Vendor ID:	0x0483  (STMicroelectronics)
  Version:	2.00
  Serial Number:	00000000001B
  Speed:	Up to 12 Mb/sec
  Manufacturer:	FrSky
  Location ID:	0x14100000 / 19
  Current Available (mA):	500
  Current Required (mA):	100

I connected then with:
screen -L /dev/tty.usbmodem1411

projectkk2glider, 2015-07-30:
Cool, so MAC and Linux handle this natively, Windows XP - 8.1 we provide driver from ST, Win 10 (I've read) has its own driver for this type of USB devices.

The speed setting is irrelevant at this point.

raphaelcoeffic, 2015-07-30:
Yes, and it would be just great if I could use it as a debug output instead of my FTDI-basic plug. Should I figure that out once you're done, or do you want to take it?

kilrah, 2015-07-31:
> Win 10 (I've read) has its own driver for this type of USB devices.

Looks like it will need the ST driver as well. 

[image]

projectkk2glider, 2015-07-31:
I know why it is not working. Taranis has in its USB device descriptor both bDeviceClass and bDeviceSubClass set to zero, so each implemented interface can then use different class/subclass.

But according to https://msdn.microsoft.com/en-us/library/windows/hardware/dn707976%28v=vs.85%29.aspx the Win10 only loads generic driver for devices that set the class code to 02 and subclass code to 02.

bsongis, 2015-08-11:
**>help**
beep [<frequency>] [<duration>]
ls <directory>
play <filename>
print <address> [<size>] | <what>
trace on | off
volume <level>
help [<command>]
**>beep 1000 100**
**>ls /SOUNDS**
.
..
en
fr
**>play /SOUNDS/fr/SYSTEM/tada.wav**
**>print RCC**
DUMP 136 bytes ...
83 6F 03 03 0C 3C 40 05 0A 94 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 1F 00 60 00 80 00 00 00 00 00 00 00 00 00 00 00
B6 C1 26 30 23 41 02 00 00 00 00 00 00 00 00 00 FF 91 67 7E F1 00 00 00 01 00 00 00 00 00 00 00
FF C9 FE 36 33 5F 07 00 00 00 00 00 00 00 00 00 03 81 00 00 02 00 00 0E 00 00 00 00 00 00 00 00
00 00 00 00 00 30 00 20
**>print USART1**
DUMP 28 bytes ...
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

closed 2015-08-31 by projectkk2glider

xros, 2015-10-28:
@raphaelcoeffic I haven't gotten a clue about connecting to Taranis  X9D Plus on Linux. The serial device is /dev/ttyACM0 and I used cutecom(GUI) to connect to it at baud rate 9600. But I could not get what @bsongis showed in the upper comment scope after typing "help". The firmware is 2.1.3 and system ubuntu 14.04 AMD64. I'm newbie to STM32 stuff. Someone can help? Thanks!

bsongis, 2015-10-28:
help gives the list of commands recognized

kilrah, 2015-10-28:
I never managed to get an answer there either. I can open the port, but never got any prompt or reply to any command.

projectkk2glider, 2015-10-28:
There were some related recent fixes like 637d20a66d82d7be39c482c02b3981599e2122df and 48a11f63ea82851542d36805fe72c9c3b8128bef. That fixed debug output trough the serial port in the battery bay. Perhaps something else similar is broken for USB interface.

I will try this later today on my radio and report back.

xros, 2015-10-28:
@kilrah  Me too. Being able to open the port but no prompt out...

projectkk2glider, 2015-10-28:
Problem confirmed, even debug prints over serial port in battery bay are not working.

projectkk2glider, 2015-10-28:
Wait a minute, ignore my previous comment. On development 9XE board it works (with current master  6f30718745e0a495e509e8ee907b402f486009c1) on battery bay serial port. One has to:
* use baudrate 115200
* compile with `CLI=YES`
* set `Serial port` in HARDWARE page to `Debug` -< this is **important** otherwise it won't work
* use command `trace on` to enable debug prints

Now I am going to test the USB serial port...

kilrah, 2015-10-28:
> set Serial port in HARDWARE page to Debug -< this is important otherwise it won't work

Aha, then that's where the problem is, especially as Debug is not available on public builds.
This should only apply to the serial port ( as it is shared between various functions), not to the CLI.

projectkk2glider, 2015-10-28:
Agreed, let me see if I can fix it. But that setting is only important for battery bay serial port. For USB serial it is not imortant.

You might also have another problem. I have just tried CLI via USB and it works, but the NetworkManager process is trying to access the port thinking it is a modem. To disable it you add UDEV rule like this to file `/etc/udev/rules.d/46-taranis-serial-port.rules`:
```
ACTION!="add|change", GOTO="mm_usb_device_blacklist_end"
SUBSYSTEM!="usb", GOTO="mm_usb_device_blacklist_end"
ENV{DEVTYPE}!="usb_device",  GOTO="mm_usb_device_blacklist_end"

ATTRS{idVendor}=="0483", ATTRS{idProduct}=="5740", ENV{ID_MM_DEVICE_IGNORE}="1"
```

I will now try USB serial on the Taranis.

projectkk2glider, 2015-10-28:
Oh another thing the 2.1.3 **wont work** because it is missing those two commits I mentioned above. So one has to use current master.

@xros where did you get 2.1.3 that has USB serial and CLI enabled? That version will not work.

projectkk2glider, 2015-10-28:
Also works on my Taranis. Question for everybody who is having problems:
* show us the make command line
* on linux: before plugging in USB cable issue command in terminal:
```
tail -f /var/log/syslog
```
and  then plug in the USB, what do you get. My example:
```
Oct 28 17:49:36 mambo kernel: [249487.021500] usb 2-1.8: new full-speed USB device number 80 using ehci-pci
Oct 28 17:49:36 mambo kernel: [249487.115678] usb 2-1.8: New USB device found, idVendor=0483, idProduct=5740
Oct 28 17:49:36 mambo kernel: [249487.115684] usb 2-1.8: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Oct 28 17:49:36 mambo kernel: [249487.115687] usb 2-1.8: Product: FrSky Taranis Serial Port
Oct 28 17:49:36 mambo kernel: [249487.115689] usb 2-1.8: Manufacturer: FrSky
Oct 28 17:49:36 mambo kernel: [249487.115691] usb 2-1.8: SerialNumber: 00000000001B
Oct 28 17:49:36 mambo kernel: [249487.115992] cdc_acm 2-1.8:1.0: This device cannot do calls on its own. It is not a modem.
Oct 28 17:49:36 mambo kernel: [249487.116016] cdc_acm 2-1.8:1.0: ttyACM0: USB ACM device
Oct 28 17:49:36 mambo mtp-probe: checking bus 2, device 80: "/sys/devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.8"
Oct 28 17:49:36 mambo mtp-probe: bus: 2, device: 80 was not an MTP device
```

One can see the radio introducing itself as `FrSky Taranis Serial Port` and becoming device `/dev/ttyACM0`

Then use miniterm.pt (comes with Python) to communicate:
```
 miniterm.py --lf /dev/ttyACM0 9600
```
After first press of Enter key the radio should respond with a prompt sign `>`, then command `help` will print out short help, etc...
Example:
```
$ miniterm.py --lf /dev/ttyACM0 9600
--- Miniterm on /dev/ttyACM0: 9600,8,N,1 ---
--- Quit: Ctrl+]  |  Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

>help
beep [<frequency>] [<duration>]
ls <directory>
play <filename>
print <address> [<size>] | <what>
stackinfo <tid>
trace on | off
volume <level>
help [<command>]
>trace on
>eeprom write general
referenceModelAudioFiles(): using file: AltHold-ON.wav
	found: AltHold-on.wav
referenceModelAudioFiles(): using file: Stabilize-ON.wav
	found: Stabilize-on.wav
referenceModelAudioFiles(): using file: RTL-on.wav
	found: RTL-on.wav
referenceModelAudioFiles(): using file: PosHold-ON.wav
	found: PosHold-on.wav
referenceModelAudioFiles(): using file: Auto-ON.wav
	found: Auto-on.wav
referenceModelAudioFiles(): using file: Loiter.wav

```

kilrah, 2015-10-28:
OK, tried with master right now on Taranis+ on Win, and it works fine. 

It however does NOT work with the latest nightly with cli option ticked in companion. Something might be missing in the compile script.

projectkk2glider, 2015-10-28:
Yes on compilation server the cli option is only turning on USB=SERIAL, we forgot CLI=YES.

projectkk2glider, 2015-10-28:
@kilrah  you only have a write permission for getfw.php!

projectkk2glider, 2015-10-28:
The current getfw.php script does not have a way to set two compile options with one Companion option. The `cli` option should set `USB=SERIAL CLI=YES`

...unless we get a litle adveturous and use:
```php
"cli"=>array("USB", "SERIAL CLI=YES", ""),
```

bsongis, 2015-10-28:
Yes it will be good until the next complete rewrite of this script, I would like we add the options in the repository, not in this script ...

kilrah, 2015-10-28:
OK, with that hack it works, even on 2.1.3... so it was just the missing CLI=YES.

I updated the script on 2.1 release and 2.1 nightly, and deleted anything that included CLI from the cache so further downloads should work.

bsongis, 2015-10-28:
Did you add write permissions to the opentx group?

xros, 2015-10-28:
@projectkk2glider I got the 2.1.3 firmware directly from companion 2.1.3. I selected the **cli** in *Build Options*. If you were right, I guess that would miss another building option -- *USB Serial*.

And this is what I got using `tail -f /var/log/syslog`
> Oct 29 04:57:30 universe kernel: [285066.680576] usb 2-1.2: new full-speed USB device number 29 using ehci-pci
Oct 29 04:57:30 universe kernel: [285066.775373] usb 2-1.2: New USB device found, idVendor=0483, idProduct=5740
Oct 29 04:57:30 universe kernel: [285066.775381] usb 2-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Oct 29 04:57:30 universe kernel: [285066.775386] usb 2-1.2: Product: FrSky Taranis Serial Port
Oct 29 04:57:30 universe kernel: [285066.775389] usb 2-1.2: Manufacturer: FrSky
Oct 29 04:57:30 universe kernel: [285066.775393] usb 2-1.2: SerialNumber: 00000000001B
Oct 29 04:57:30 universe kernel: [285066.775814] cdc_acm 2-1.2:1.0: This device cannot do calls on its own. It is not a modem.
Oct 29 04:57:30 universe kernel: [285066.775847] cdc_acm 2-1.2:1.0: ttyACM0: USB ACM device
Oct 29 04:57:30 universe mtp-probe: checking bus 2, device 29: "/sys/devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.2"
Oct 29 04:57:30 universe mtp-probe: bus: 2, device: 29 was not an MTP device
Oct 29 04:57:34 universe kernel: [285070.719210] usb 2-1.2: USB disconnect, device number 29
Oct 29 04:57:34 universe ModemManager[881]: <info>  (tty/ttyACM0): released by modem /sys/devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.2
Oct 29 04:57:34 universe ModemManager[881]: <warn>  (Plugin Manager) (Via CBP7) [ttyACM0] error when checking support: '(Via CBP7) Missing port probe for port (tty/ttyACM0)'
Oct 29 04:57:34 universe ModemManager[881]: <warn>  (Plugin Manager) (Iridium) [ttyACM0] error when checking support: '(Iridium) Missing port probe for port (tty/ttyACM0)'
Oct 29 04:57:34 universe ModemManager[881]: <warn>  (Plugin Manager) (Cinterion) [ttyACM0] error when checking support: '(Cinterion) Missing port probe for port (tty/ttyACM0)'
Oct 29 04:57:34 universe ModemManager[881]: <warn>  (Plugin Manager) (Generic) [ttyACM0] error when checking support: '(Generic) Missing port probe for port (tty/ttyACM0)'
Oct 29 04:57:34 universe ModemManager[881]: <warn>  Couldn't find support for device at '/sys/devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.2': not supported by any plugin
Oct 29 04:57:36 universe kernel: [285072.284053] usb 2-1.2: new full-speed USB device number 30 using ehci-pci
Oct 29 04:57:36 universe kernel: [285072.379364] usb 2-1.2: New USB device found, idVendor=0483, idProduct=5740
Oct 29 04:57:36 universe kernel: [285072.379372] usb 2-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Oct 29 04:57:36 universe kernel: [285072.379376] usb 2-1.2: Product: FrSky Taranis Serial Port
Oct 29 04:57:36 universe kernel: [285072.379380] usb 2-1.2: Manufacturer: FrSky
Oct 29 04:57:36 universe kernel: [285072.379384] usb 2-1.2: SerialNumber: 00000000001B
Oct 29 04:57:36 universe kernel: [285072.379993] cdc_acm 2-1.2:1.0: This device cannot do calls on its own. It is not a modem.
Oct 29 04:57:36 universe kernel: [285072.380035] cdc_acm 2-1.2:1.0: ttyACM0: USB ACM device
Oct 29 04:57:36 universe mtp-probe: checking bus 2, device 30: "/sys/devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.2"
Oct 29 04:57:36 universe mtp-probe: bus: 2, device: 30 was not an MTP device
Oct 29 04:57:42 universe wpa_supplicant[1293]: wlan0: CTRL-EVENT-SCAN-STARTED 
Oct 29 04:57:47 universe wpa_supplicant[1293]: nl80211: send_and_recv->nl_recvmsgs failed: -33                                                                                      
Oct 29 04:57:54 universe ModemManager[881]: <info>  Creating modem with plugin 'Generic' and '1' ports                                                                              
Oct 29 04:57:54 universe ModemManager[881]: <warn>  Could not grab port (tty/ttyACM0): 'Cannot add port 'tty/ttyACM0', unhandled serial type'                                       
Oct 29 04:57:54 universe ModemManager[881]: <warn>  Couldn't create modem for device at '/sys/devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.2': Failed to find primary AT port       
                                                                                                                                                                                    
                                                                                                                                                                                    
                                 

I might have the same output as you had.

So with the compiling options `USB=SERIAL CLI=YES`, can it work now?

I would assume so but have not compiled one firmware like that yet.

projectkk2glider, 2015-10-28:
@xros  we had a bug on compilation server, Kilrah has fixed it now. So please re-download OpenTX firmware from the Companion and flash radio. It should work now.

xros, 2015-10-28:
@projectkk2glider  it works now. This is marvelous! Many applause to all! Thank you!

I have gotten into the debug mode now. Great job!

kilrah, 2015-10-29:
> Did you add write permissions to the opentx group?

Hadn't, but done now.

kilrah, 2015-10-30:
Hmm I now see that this actually defines DEBUG, and according to @projectkk2glider 's comment on RCG it should probably be made obvious that it shouldn't be used for flying, maybe by renaming the "cli" option to "debug" in companion.

projectkk2glider, 2015-10-30:
>it should probably be made obvious that it shouldn't be used for flying

:+1:

commit 2b312fdb (2015-08-31): Fixes #2595 - Cli added

Based on projectkk2glider work, with a lot of bugs I am sure. Still not
tested, this is why I preferred working on my own branch. I push it now
so that Jenkins do the first tests and you can review if I am on the
right track. 

I am really wondering if a merge with the Lua api would be possible /
nice, what do you think?
commit 771206e1 (2015-08-31): Merge pull request #2696 from opentx/bsongis/issue2595_cli

Fixes #2595 and #2598: USB serial port and CLI added