opentx/opentx #2596: Companion 2.1 writing bad eeprom files to mega 2560

WillD7545, 2015-07-29:
I thought it was a problem with the board, so I've acquired a second board. One has 16u2, the other has 340G. Both end up with "bad eeprom data" on radio when trying to write to boards. I've tried uninstalling completely and deleting program and temp files and re-installing, this didn't work.

JivaroFAD, 2015-07-29:
Please could you confirm that you flash the right fuses and the bootloard provided in http://jenkins.open-tx.org/MEGA2560/2_Configuration-Flash/ ?

What is the origin of the eeprom that you try to write ? Could you give it to us ?

WillD7545, 2015-07-29:
I flashed the provided fuses and bootloader from the OpenTX website. I'm using an eBay uabasp programmer.

WillD7545, 2015-07-29:
And the eeprom I'm trying to write is just models that I create in companion.

JivaroFAD, 2015-07-30:
Please could you try the following procedure :
- configure Companion with M2560 radio profile and usb STK500 connection (not usbasp programmer)
- download M2560 firmware
- connect M2560 to usb PC
- flash M2560 board (choose : no back-up of eeprom)
- let M2560 format eeprom
- disconnect M2560 then reconnect to usb PC
- download eeprom to Companion
- make modification of eeprom on Companion
- disconnect M2560 then reconnect to usb PC
- write this eeprom to M2560

WillD7545, 2015-07-30:
That's how I do it and it ends up with "bad eeprom data".

tmrttmrt, 2015-09-03:
Writing eeprom with selecting Check Hardware compatibiliy tick box appears to be working OK.

bsongis, 2015-09-03:
I think I fixed it already, this bug was also present on 9XR-PRO

tmrttmrt, 2015-09-03:
Version 2.1.1 of Companion from  August the 6th still has this issue.

tmrttmrt, 2015-09-03:
Current master appers OK in this respect indeed.

tmrttmrt, 2015-09-03:
Sorry, I was wrong. In 2.1.2N439 the issue is still here for  MEGA2560. 

Steps to reproduce the problem:

1. Read Models and Settings from MEGA2560.
2. modify some model settings.
3. Write Models and Settings to Radio with  Check Hardware compatibiliy tick box unchecked.
4. Bad EEprom Data.
5. repeat Write Models and Settings to Radio with  Check Hardware compatibiliy tick box checked.
6. EEprom OK.

Everything done using WIRING programmer option in AVRDUDE configuration to prevent blocking.

gtraveler, 2016-11-02:
Is there a workaround upload an eeprom while the bug is being fixed? Thanks!!

StefoTheHighElf, 2018-02-22:
Same issue. Steps:
- fuses and bootloader upload
- disconnecting the board from ICSP and connect the usb
- uploading firmware and thats the last point I can communicate with the board, unless reflash bootloader. Next try to upload or read from the board ends with timeout

ALFicek84, 2018-02-26:
same issue here

1) cant untick "backup of setting/models" (no back-up of eeprom) in companion 2.2.1
[image]

2) when i try to flash FW to clean mega2560 board (write fuses, write bootloader etc.), companion reads and writes just backups and settings, no FW loading at all (STK500 mode thru USB). I use AVRDude instead write FW and im ending with STK500 timeout when i try to sync with companion. Using sync in usbasp mode seems to work.

click write FW to TX and output is

C:/Program Files (x86)/OpenTX/Companion 2.2/avrdude.exe -c stk500v2 -p m2560 -D - F -P com15 -U flash:r:C:/Users/lu062680/AppData/Local/Temp/3612-flash-check.bin:r

=================================================================================

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude.exe: Device signature = 0x1e9801

avrdude.exe: reading flash memory:

Reading | ################################################## | 100% 29.71s

avrdude.exe: writing output file "C:/Users/lu062680/AppData/Local/Temp/3612-flash-check.bin"

avrdude.exe done. Thank you.

=================================================================================

C:/Program Files (x86)/OpenTX/Companion 2.2/avrdude.exe -c stk500v2 -p m2560 -D - F -P com15 -U eeprom:r:/backup-2018-02-24-173147.bin:r

=================================================================================

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude.exe: Device signature = 0x1e9801

avrdude.exe: reading eeprom memory:

Reading | ################################################## | 100% 0.46s

avrdude.exe: writing output file "/backup-2018-02-24-173147.bin"

avrdude.exe done. Thank you.

=================================================================================

closed 2020-05-01 by kilrah