opentx/opentx #2598: Add USB serial interface

projectkk2glider, 2015-07-29:
Related issues #2008 and #2595.

I will take a stab at implementing this now.

projectkk2glider, 2015-07-29:
Some resources for this issue:

Composite USB device:
 * https://my.st.com/public/STe2ecommunities/mcu/Lists/cortex_mx_stm32/Flat.aspx?RootFolder=%2Fpublic%2FSTe2ecommunities%2Fmcu%2FLists%2Fcortex_mx_stm32%2FMultiple%20USB%20CDC%20%28USB%20IAD%29.%20How&FolderCTID=0x01200200770978C69A1141439FE559EB459D7580009C4E14902C3CDE46A77F0FFD06506F5B&currentviews=6903
 * http://dccharacter.blogspot.com/2013/04/using-stm32-usb-fs-device-development.html

Some general USB resources:
 * http://www.usbmadesimple.co.uk/index.html
 * http://www.beyondlogic.org/usbnutshell/usb1.shtml

projectkk2glider, 2015-07-29:
Update on my progress: I have a working example that just returns all received characters. But it is still messy and not ready for commit.

bsongis, 2015-07-29:
Cool!

projectkk2glider, 2015-07-29:
This is just for my reference.

Udev rule needed to stop ModemManager from trying to open Taranis serial port:
```
$ cat /lib/udev/rules.d/45-companion21-taranis.rules
ACTION!="add|change", GOTO="mm_usb_device_blacklist_end"
SUBSYSTEM!="usb", GOTO="mm_usb_device_blacklist_end"
ENV{DEVTYPE}!="usb_device",  GOTO="mm_usb_device_blacklist_end"

ATTRS{idVendor}=="0483", ATTRS{idProduct}=="5710", ENV{ID_MM_DEVICE_IGNORE}="1"
```

PR #2682 (2015-08-31): Fixes #2598: added USB serial port 
Closes #2598 

Use `USB=SERIAL` to enable. If `DEBUG=YES` then debug output also goes to the USB serial port.

The CLI is not implemented, the current example only returns all received characters with a `+` character injected between each one. The buffer sizes should also be tuned.

The USB peripheral is now initialized and ready to go from the start. The reason is stated in the comments. There is now a slight chance that this could affect the radio operation in a negative way even with the USB cable unconnected. But I think it will be OK.
commit f7a87156 (2015-08-31): Fixes #2598: added USB serial port (with echo example), use USB=SERIAL to enable. If DEBUG=YES then debug output also goes to the USB serial port
closed 2015-08-31 by projectkk2glider