opentx/opentx #2601: Play Value Battery cant play decimal number

ChadRR, 2015-07-29:
Is the play value bat dont play decimal number? Or is it a bug?

projectkk2glider, 2015-07-29:
Is this the same issue as #2545?

ChadRR, 2015-07-29:
Nope, its in Tx. I would like to play the value of the battery in Tx. But i think that will do. Thanks

kilrah, 2015-07-29:
Radio battery should play the decimal correctly.

Make sure you have the correct voice pack for your firmware version.

ChadRR, 2015-07-29:
I have the latest firmware.

ChadRR, 2015-07-29:
Ill download it tomorow. I got it thanks again.

closed 2015-07-29 by ChadRR

projectkk2glider, 2015-07-29:
It is not a problem in your firmware, the problem is in your language pack! The 2.1 version has different files for voice reproduction. With sounds from 2.0 the this issue will emerge. With sounds for 2.1 this issue is non-existent. 

Download new sound files via Companion Download dialog.

kilrah, 2015-07-29:
> I have the latest firmware.

I wish people would stop saying this... Firstly because of so many users that say it when they didn't actually check and there's something newer they didn't notice, that often solves their issue but gets us look for nothing. 
And more importantly when like in this case we currently actually maintain 2 branches that both have a valid "latest version" but we now have no idea of which of them you're talking about.

Sorry for the rant.

ChadRR, 2015-07-30:
ok, im so sorry. and thank you again.