opentx/opentx #2603: Optionally start simulator in "debug mode" where it has more resources available than on the radio.

fraternl, 2015-07-30:
Adrian has written a launch script that helps with practicing launches.
http://www.rcgroups.com/forums/showthread.php?t=2467581

Alas, it gives a blank screen in OpenTX 2.1.x

It also does this in the simulator, which in itself is of course a good thing.

But I would think the simulator would have much more resources available and would therefore run these LUA-scripts without any problem.
If it would be possible to optionally change the amount of resources available for the simulator, the program could be debugged and changed, so it uses less resources.

projectkk2glider, 2015-07-30:
There is no such limitation in simulator. Have you looked at the debug output (F6 key in simulator)? I bet the problem is somewhere else.

closed 2015-07-30 by projectkk2glider

fraternl, 2015-07-30:
Thanks for the info....

fraternl, 2015-07-31:
@projectkk2glider 

Hi Damjan,

The info about the debugger in the simulator helped me to find the culprit, but as I'm not a seasoned LUA-scripter for OpenTX I'm using my intuition to change things.

It seems the interpreter doesn't like it when system functions are used directly in formulas.
Things seem to work better if I put it in a local variable first....  Is this true? If so, is this intended?

I'm not doing the programming for this project: http://www.rcgroups.com/forums/showthread.php?t=2467581

But I'm trying to help.

If you have time, could you read through that thread?
I'm making assumptions there and would like to be corrected if I'm wrong..

projectkk2glider, 2015-07-31:
I have no time to debug that script, but there is #2595 in development that should help Lua developers in such cases. 

We will soon have the USB serial port emulation and when connected to the terminal program in PC you will be able to see the debug output from the radio in the terminal.  That should work the same as in Companion simulator (F6 debug window) but with the difference of running in real radio (different memory limits). You will also be able to see Lua memory usage and a bunch more.

fraternl, 2015-07-31:
That wasn't my intention for you to debug the script....
I just wanted to know if it's correct we can't use system functions in formulas and we have to put them in variables first....
But I understand you all are extremely busy....

BTW....   Isn't it a good idea to make the current builds a released version (2.1.1)? 
I know some things still needs to be done, but there are a few nasty things in 2.1.0 that would hold off people from using it.

projectkk2glider, 2015-07-31:
>I just wanted to know if it's correct we can't use system functions in formulas and we have to put them in variables first....

There in no such limitation, the OpenTX Lua functions should behave just like the other Lua functions. 

The 2.1.1 is planned to be released over the week-end.