opentx/opentx #2604: Feature: Real-time updates

JoeB99, 2015-07-30:
I'd like to use companion to configure my taranis models, changing settings on the pc and clicking a "send to taranis" button, and then the taranis will accept these new changes as if i'd changed them on the taranis without having to enter and exit boot mode each time.

This would be useful when, for example, setting up a new model on the bench, for servo reversing, trimming etc. 

At the moment, i set up model on companion, send to taranis via boot mode, then use the radio menus to trim, reverse servos, travel adjust etc, then when i'm happy, i upload back to the pc.

projectkk2glider, 2015-07-30:
Similar/related to #2008

geekness, 2017-01-13:
I know it's an old thread, but any idea if this feature has progressed at all?

projectkk2glider, 2017-01-13:
Nope.

closed 2020-07-07 by stale[bot]