opentx/opentx #2605: Writing setting to radio doesn't work -  Companion 2.1 /  9XR PRO

procha170, 2015-07-30:
Hello,
Save setting doesn't work in companion. I can load setting from radio but changes are no written there.  9XR PRO

projectkk2glider, 2015-07-30:
Same as #2517?

procha170, 2015-07-30:
Yes It looks the same. Official version 2.1 freeze during saving. Last build doesn’t freeze but saves nothing.

projectkk2glider, 2015-08-02:
@procha170 please check again after the 2.1.1 is released.

kilrah, 2015-08-02:
Possibly related to #2619...

bsongis, 2015-08-03:
I am afraid that the bootloader doesnt accept the new eeprom format.
Here is the code in the sky9x bootloader:
'''
// Test for EEPROM file
uint32_t isEepromStart( uint8_t *p )
{
	uint32_t csum ;
	uint32_t size = 7 ;

	csum = 0 ;
	while( size )
	{
		csum += *p++ ;
		size -= 1 ;
	}
	if ( csum == *p )
	{
		return 1 ;
	}
	return 0 ;	
}
'''

Fortunately the result of the test goes into a static boolean which never returns to 0, it should be possible to bypass the test.

'''
if ( isEepromStart( (uint8_t *) buffer ) )
		{
			EepromBlocked = 0 ;
		}
		else
		{
			EepromBlocked = 1 ;
		}
'''

kilrah, 2015-08-03:
We should port our bootloader at some point...

commit 6e6e80bc (2015-08-04): Fixes #2605
closed 2015-08-04 by bsongis