opentx/opentx #2607: VFAS ID 0x39 Selectable precision

flarssen, 2015-07-31:
ref #2531
Can the interpretation of this value be based on the selected precision for the sensor (default; 0.1V)?
This would give support for both 0.1V and 0.01V sensors.

projectkk2glider, 2015-08-02:
An idea how to implement this:

The current 0x39 D type sensor sends voltage in int16_t variable with a resolution of 0.1V. The possible range is  -3276.8V to 3276.7V which is much bigger than needed. Actually the negative range is not needed at all.

So lets define two ranges:
* if incoming value is in a range from 0 to 2000 we treat it as before, the resolution is 0.1V and available range is from 0.0V to 200.0V
* if incoming value is bigger than 2000, then subtract 2000 from it and treat it as a value with a resolution of 0.01V. Available range in this case if from 0.00V to approx 300.00V

Example:
* incoming value 4567
* bigger than 2000, substract -> 4567 - 2000 = 2567
* the voltage is 25.67V

Should be very easy to implement in OpenTX and in any custom sensor. Existing sensors should not be affected (unless they are currently sending values bigger than 200 volts).

kilrah, 2015-08-02:
That may not work correctly or require dirty hacks in OpenTX as the expected received precision of a given ID is hardcoded.

projectkk2glider, 2015-08-02:
@kilrah  This is my idea 121f7780c2db5838fb8f824c9bcc03036e4250cf, not tested, but it should work and not break existing stuff.

kilrah, 2015-08-02:
Indeed why not, the 2nd decimal is shown for nothing in "low precision" mode but that doesn't hurt. Was already the case.

PR #2622 (2015-08-03): Fixes #2607: Added support for 0.01V resolution to D series VFAS ID 0x39
Closes #2607 

Tested in simu with the help of gdb (manually calling function `call processHubPacket(0x39, 14087)`) and it seems to work.
commit 121f7780 (2015-08-03): Fixes #2607: Added support for 0.01V resolution to D series VFAS ID 0x39
commit 5417c426 (2015-08-03): Merge pull request #2622 from opentx/projectkk2glider/issue_2607

Fixes #2607: Added support for 0.01V resolution to D series VFAS ID 0x39
closed 2015-08-03 by bsongis