opentx/opentx #2611: Move to Qt 5

bsongis, 2015-08-01:


griebd, 2015-08-01:
on the topic of upgrading development tools...
I'm still using gcc 4.8 with no problems so far!

$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.8.4 20140725 (release) [ARM/embedded-4_8-branch revision 213147]

For companion, using cmake -DCMAKE_BUILD_TYPE=Release and gcc 4.8, I have no problems except by some warnings (because -Wstrict-aliasing and -Warray-bounds) at compile time.

$ gcc --version
gcc (GCC) 4.8.2

raphaelcoeffic, 2015-08-01:
And Qt5 has support for retina displays :-)

amontefusco, 2015-08-06:
Hi Bertrand,
I have just rebuilt on Ubuntu 14.04 both companion21 and simulator21 using Qt5 and Phonon4-qt5.
If you find it useful, I will create my fork  with all the changes in  qt5 branch.

bsongis, 2015-08-06:
Why didn't tell it before?
I have it stashed here, with Qt5Multimedia instead of Phonon

amontefusco, 2015-08-06:
I started to work on it today in the afternoon and finished one hour ago :-)

bsongis, 2015-08-06:
Ok then we will take the merge of both :)
I removed a lot of includes
I am thinking about switching everything to CMake as well ...

amontefusco, 2015-08-07:
https://github.com/amontefusco/opentx/

Sorry for the delay but I am currently in val Badia and there was an Internet outage just past night.
In the meantime, I fixed a bug (no pulldown menu).

bsongis, 2016-03-10:
Done!

closed 2016-03-10 by bsongis

kilrah, 2016-03-10:
Cool!