opentx/opentx #2615: Improve use of "ratio" parameter in telemetry sensors

kilrah, 2015-08-01:
"Ratio" currently has the same effect as the "range" parameter for analog sensors on pre-2.1, i.e. "how much the output should be when the input is 255".

It is still required for analog sensors nowadays (yet could be improved for S.port type that have a full scale of 330 instead of 255), but it unfortunately makes no sense at all for all other telemetry sources.

The use of that parameter needs to be differentiated between analog sensors (no change) and the others (a simple multiplying factor, with PREC2 so as to allow for pretty precise conversion factors both when >1 and <1.

frobert0, 2015-12-06:
Ok works, only the determination of the required value is difficult.

In time was 43 instead of 50

Set I was ordered 25 ????????????????

kilrah, 2015-12-06:
This issue is precisely about making that better, for now the number is "arbitrary".

projectkk2glider, 2016-05-20:
I think 2.2 would be a better target for this - it might require an EEPROM change.

closed 2020-04-06 by stale[bot]