opentx/opentx #2617: Various problems with the Vario settings and announcements after  upgrade to opentx 2.1.0

AnthonyPeate, 2015-08-02:
I have just upgrade my Taranis to opentx-taranis-2.1.0 so I could use the new vario silence band feature. The openTx EEPROM version show after the upgrade in the radio was 217.

I also updated the Companion 9X to the latest version prior to upgrading the Taranis to 2.1.0

All of the problems listed below were observed after the upgrade. 

Problem #1 
1. Added Global Function GF1 - SG(down) - Play Value - Alt - 45s 
2. Put switch SG in the down postilion.
3. Connected high precision FRSky Vario to SBus on X8R received and powered up

I expected with switch SG down the following announcement every 45 seconds: {Alt value}{feet}. However {ALt value}{m-a-h} was announced.

My Telemetry Page Vario set up was as follows:-
Sensor 5
Name: Alt
Type: Custom
ID: 0100 0
Unit: ft
Precission: PREC0
Ratio: -
Ofset: 0
Auto Offset: Ticked
Filter: Ticked
Logs: Ticked

Sensor 6
Name: VSpd
Type: Custom
ID: 0110 1
Unit: f/s
Precission: PREC0
Ratio: -
Ofset: 0
Auto Offset: Ticked
Filter: Ticked
Logs: Ticked

Variometer
Source: VSpd
Range:  -15    25
Center: -2.1   0.1   Silent

Problem #2:
When using Global Functions as follows:-
GF7  {Any Switch} - Play Track - {select any announcement}
The announcement fails to select and "---" is shown

Problem #3:
There doesn't seem to be a system announcement for "feet per second" to be appended to the VSpd Play Value when the Vario is in f/s mode as per the example above. Perhaps a new 0170.wav file is needed?

Question #1:
if the Alt is set to Unit "feet" or the VSpd is set to Unit "f/s" do I need to change the Ratio value setting to convert from meters to feet?

Question #2:
Also notice that one glider I have starts on Alt ~= 17000 feet Alt instead of 0 feet at power up unless tick the Auto Offset in the Sensor 5 Alt settings however another glider with the same type of FR Sky High precision vario start at 0 feet without ticking Auto Offset. Both Gliders use the X8R and SBus. This is a bit odd.Have the harder variants of the FRSky High precision Vario changed?

In Summary.

I'd like to thank the clever guys in the opentx development team for providing such a great radio system to the model glider pilots at the VARMS (Victorian Aero Modelling Soaring Society) in Melbourne Australia. This radio system is now very popular because of it's flexibility and the excellent telemetry data system. 

If these few remaining bugs can be addressed for the Vario functions, I believe this radio system will be the best by far for model glider pilots.

I hope this feedback has been useful in making this great TX even better.

griebd, 2015-08-02:
Some of your problems are apparently related to sound files... Have you upgraded your sound files? They have changed from 2.0 to 2.1!

kilrah, 2015-08-02:
The problems are either related to you needing to update your sound pack, or are already fixed for 2.1.1.

Question 1, no you just change the unit in the sensor settings. Question 2 no idea...

closed 2015-08-02 by kilrah

RidgeMan, 2016-08-24:
i'm having the same problem... did anyone figure out the solution?

tomkfly, 2016-08-24:
The solution is posted, in answer to similar questions, daily on RC groups and also by Kilrah in the post above yours. "You have the wrong sound pack"

RidgeMan, 2016-08-24:
I will try to reload the newest amber pack I can find or just go back to the newest standard sound pack I suppose...  would you have any links to the newest packs out there? 

thanks much for the help! RidgeMan

3djc, 2016-08-24:
Offical voices for 2.1 can be found here : http://downloads.open-tx.org/2.1/voices/

RidgeMan, 2016-08-24:
yeah... it was pointed out to me in RCG... I just downloaded the amber pack
as posted here:  http://open-txu.org/home/v2-1-resources/ ... and compared
to the sounds on my sd.. they are different... I truly thought that I did
this before... I guess I screwed up... sorry for wasting everyones time!

thanks again for the help!

AnthonyPeate, 2016-09-04:
Thanks guys. Yes down loading the new sound file pack fixed most of my issues listed above.

I still can't use gloable functions to assign vario or altitude related play of announcements to switches that would be common to all gliders. I have had to set this on each glider individually. I have also found i need to use high precision vario to get any useful vertical speed and thermal tones.