opentx/opentx #2621: LUA doesn't have enough memory after switching models

fraternl, 2015-08-02:
"LUA doesn't have enough memory after switching models"
I don't know if it really is true, but the symptoms point in that direction.

I am running 2 models that are using this launch script.

The setup is nearly identical and they both run fine after starting the radio.
http://www.rcgroups.com/forums/showthread.php?t=2467581
The LUA-script fails however when I switch from one model to the other.
This may be an indication that memory isn't (fully) returned to the system.

Switching models in the simulator is no problem

kilrah, 2015-08-02:
firmware version?

fraternl, 2015-08-02:
Current nightlies
2.1.1N390

projectkk2glider, 2015-08-05:
This will be easier to debug when we have #2598. 

In the meantime I would like to see that script in a github repository where I could add some general comments on how to optimize it to use less resources. I already saw a couple of improvements after glancing over the script. @fraternl  try to get the author to create a repository on github for it.

fraternl, 2015-08-05:
I can try and already did that a while ago.
Would be great if this became a project where people can contribute in a structured way

projectkk2glider, 2015-08-09:
Just for reference.

I am playing with this script (lm114jp.lua) on 9XE and monitoring Lua memory usage with the #2682, which has more RAM available for Lua scripts. So I don't get the original issue when changing models. In fact the memory usage by Lua is the same each time I change the models.

But another interesting observation. When compiled with:
```
make clean; make  PCB=TARANIS  PCBREV=REV9E USB=SERIAL DEBUG=YES LUA=YES
```
I get Lua usage `GC Use: 43089bytes`

If I add another Lua telemetry screen with the same script (so the same script is used twice) I get immediately an out of memory error:
```
GC Use: 43067bytes
GC Use: 50105bytes
Error in script /SCRIPTS/TELEMETRY/lm114jp.lua: not enough memory
GC Use: 43089bytes
```
So only one telemetry view works, another one is blank (that is an issue in itself #2683, it should at least display some error to the user).

Now if I enable Bin Allocator:
```
make clean; make  PCB=TARANIS  PCBREV=REV9E USB=SERIAL DEBUG=YES LUA=YES USE_BIN_ALLOCATOR=YES
```
I can get the two telemetry views running simultaneously no problem:
```
GC Use: 74181bytes
```

So this proves that at least in some circumstances the Bin Allocator is beneficial. Another test was done by another user and his results were not so good https://github.com/opentx/opentx/issues/2050#issuecomment-87653572.

fraternl, 2015-09-25:
Adrian's script seems to work flawlessly, but after switching models my LCD screen becomes blank. It seems the script is not at fault. If I do the same with the SIM I don't get the same behaviour.
If the reason behind this is found it probably benefits the stability of the whole LUA environment.

fraternl, 2016-02-07:
Today I needed to turn my radio of and on again so it would run the telemetry script I always use for my DLG.

I came from another model and I got a "syntax error". That script runs flawlessly after a fresh restart.

I'm convinced some things aren't initialized properly after a model change or there's some memory leak.
Could you please test?

It's just a little inconvenience for me, but I'm afraid the source of this problem could cause more problems.

I had this same feeling when I noticed that L1 suddenly changed. This turned out to be a quite serious bug.

closed 2016-11-27 by fraternl