opentx/opentx #2623: [Bug][OpenTX 2.0.17] No refesh after deleting a file

LapinFou, 2015-08-02:
Hi folks,

From the SD CARD 2/8 menu when I'm deleting a file, it doesn't disappear from the list. I need to move to a parent folder then going back in the folder to see that it has been effectively removed.

projectkk2glider, 2015-08-04:
Tried to reproduce on simu, where I completed the simpgmspace.cpp function f_unlink() to actually delete the file:
```c
FRESULT f_unlink (const TCHAR* str)
{
  TRACE("Deleting file %s", str);
  unlink(str);
  return FR_OK;
}
```

I could not reproduce this issue, which is strange. Each time a file was deleted the contents of directory were re-read in displayed correctly without the deleted file. The functions that handle this were not changed in the last year or so!

LapinFou, 2015-08-04:
I'm able to reproduce this problem from my Taranis, not from the simulator.

projectkk2glider, 2015-08-04:
Ok, I will test this on my radio later.

projectkk2glider, 2015-08-04:
There is also a related bug. The radio allows and tries to delete the `[..]` item.

LapinFou, 2015-08-04:
Have you been able to reproduce the issue?

projectkk2glider, 2015-08-04:
Ok, reproduced. The dir must have more files than fit one on screen. The deleted file is removed from list when it scrolls out of the display. The 2.1.x is also affected.

projectkk2glider, 2015-08-08:
This one will be solved for 2.1 first and then ported back to 2.0

projectkk2glider, 2015-08-17:
This needs to be ported back to the 2.0.x branch

commit 25c7fe8c (2015-08-11): Fixes #2623: macro for node type introduced, cosmetics
PR #2691 (2015-08-17): Fixes #2623: several fixes:
Disable menu for [..],
Refresh file list after delete,
Copy destination can be a directory,
Prevent copy/paste into the same directory,
Enabled renaming of directories and files without extension,
Removed delete menu option for directories,
Several simulator filesystem functions improved

Closes #2623 

Please test thoroughly before merging!
commit 28b9a70e (2015-08-17): Fixes #2623: several fixes:
Disable menu for [..],
Refresh file list after delete,
Copy destination can be a directory,
Prevent copy/paste into the same directory,
Enabled renaming of directories and files without extension,
Removed delete menu option for directories,
Several simulator filesystem functions improved
commit 379d4323 (2015-08-17): Merge pull request #2691 from opentx/projectkk2glider/issue_2623_sd_manager_fixes

Fixes #2623: several fixes:
closed 2015-08-30 by projectkk2glider