opentx/opentx #2625: Shutdown confirmation

myckey, 2015-08-03:
When loading latest version of firmware to 9XE and selecting the shutdownconfirm option this doesn't get loaded in the firmware, i.e. just push and hold down power button shuts down the TX, no request to confirm shutdown by pressing eneter.

projectkk2glider, 2015-08-03:
Probably just a missing option on the firmware compilation server, the source looks fine.

myckey, 2015-08-03:
That would make sense as the 9XE with the shipped firmware DID have the confirmation working.

 

No virus found in this message.
Checked by AVG - www.avg.com <http://www.avg.com> 
Version: 2015.0.6086 / Virus Database: 4401/10364 - Release Date: 08/03/15

kilrah, 2015-08-03:
Typo in the compilation script, should be fixed.

closed 2015-08-03 by kilrah

myckey, 2015-08-03:
I can confirm that is now fixed, thank you.