opentx/opentx #2626: RTC Clock Resets

myckey, 2015-08-03:
On the 9XE the RTC clock resets when the battery has reached a flat state (9.6 V) and is then recharged overnight, it doen't reset if the battery is charged up from only partially discharged and I can't produce this on any other event. The RTC battery is definitely working as the clock does not reset if the main battery has been removed, the RTC reset also resets the geographic region.

kilrah, 2015-08-03:
Did you actually measure the RTC battery? A batch of radios has been shipped with flat batteries. Results are unpredictable in this state.

myckey, 2015-08-03:
I have no way of measuring it under load but it read 3.03V on a meter that is under reading slightly (just charged two cell life reads 6.52V, that is why I took out the main battery thinking that if I left it out a good few minutes that might show whether it was working or not.

 

No virus found in this message.
Checked by AVG - www.avg.com <http://www.avg.com> 
Version: 2015.0.6086 / Virus Database: 4401/10364 - Release Date: 08/03/15

myckey, 2015-08-03:
I don’t know if this sheds any light on the problem but when it resets it does so to 01-01-2010 whereas when you go to change it the earliest year is 2012.

 

In addition I just removed the RTC battery AND the main battery and the RTC resets. Removing the main battery on its own it behaves correctly and doesn’t reset.

 

I think that given the cost of a RTC battery I will get a new one when in town later on and see if I can reproduce the error with that, if I can’t then it may well be due to a dying (just not completely dead) battery though this will take me a couple of days to carry out at least twice.

 

Does that sound sensible?

 

Phil

projectkk2glider, 2015-08-05:
@myckey From you experience it seems that RTC is reset only when the radio's main battery voltage is low. It would be beneficial if you can do some more tests/observations:
* confirm that when the battery is over certain voltage (can you point approximately at which voltage it works) the RTC is kept when the radio is turned off.
* What is the voltage of the main battery when you get to a point where RTC starts resetting? 
* When you get to the point above (RTC resetting) do one test:
  * confirm that the RTC was reset.
  * set the date/time to some value (different from default)
  * turn off the radio briefly (a couple of seconds is enough)
  * turn on the radio again, check if the RTC was reset or not, if it was not, does the longer radio off time in the same conditions trigger RTC reset?

myckey, 2015-08-05:
“From you experience it seems that RTC is reset only when the radio's main battery voltage is low”.

To be exact only when it has been fully recharged from low.

*	confirm that when the battery is over certain voltage (can you point approximately at which voltage it works) the RTC is kept when the radio is turned off.

It’s never reset other than the above no matter what the battery level has got, to even right down to 9.4 volts and turning on an off has never reset it.

*	What is the voltage of the main battery when you get to a point where RTC starts resetting? 

Using default setting of 9.6 volts for alarm, taken it down to about 9.4 volts

*	When you get to the point above (RTC resetting) do one test: 

*	confirm that the RTC was reset.
*	set the date/time to some value (different from default)
*	turn off the radio briefly (a couple of seconds is enough)
*	turn on the radio again, check if the RTC was reset or not, if it was not, does the longer radio off time in the same conditions trigger RTC reset?

 

As I said I would earlier I bought a replacement RTC battery on Monday, flattened it that day and recharged overnight and RTC stayed good. Yesterday did the same, recharged it overnight and turned it on this morning, TRC was still good, thought that changing the battery had solved it, then turned it off and on quickly and the RTC reset (your points above).

 

I’m hoping that was just a fluke, the original battery wad definitely over 10% down on measured voltage compared to the new, as a lithium battery that means to me that it’s only just still functioning at all. 

 

Currently I’m taking it down to 9.9 volts and will then recharge it and can then let you know if it resets or not, I’m not expecting it to. 

 

I’m having trouble getting a repeatable result (which would lead me down the hardware rather than software being the cause).

 

I don’t know how much time you want to give this (if any) but don’t give it any on my behalf, getting a repeatable failure is difficult and takes up a lot of time and resetting the RTC is hardly a big issue and I’m prepared to live with it as the only thing I can see it affecting is the date in the telemetry logs and I can cope with that.

 

Still quite happy to try anything you want though.

 

Version: 2015.0.6086 / Virus Database: 4401/10373 - Release Date: 08/04/15

projectkk2glider, 2015-08-05:
>Yesterday did the same, recharged it overnight and turned it on this morning, TRC was still good, thought that changing the battery had solved it, then turned it off and on quickly and the RTC reset (your points above).

So if I am reading this correctly, this time RTC did a reset while the battery voltage was high (fully charged)?

myckey, 2015-08-05:
Yes that is correct, charge was cut off by the charge circuit overnight, battery voltage read 10.9 volts on the display, turned off and on quickly and it reset, updated the RTC and did the same again several times and it hasn’t reset again.

 

Just unplugged the main battery for at least half an hour then plugged it back in and the RTC is still correct so the RTC battery is definitely powering the clock.

 

Version: 2015.0.6086 / Virus Database: 4401/10375 - Release Date: 08/05/15

projectkk2glider, 2015-08-05:
So to sum it up: there is no regularity when the RTC resets, it is not related to a particular main battery voltage or to the duration of the radio power off state. This one will be interesting/challenging  to solve for sure.

myckey, 2015-08-05:
That sums it up perfectly, the only time it resets is just after (this morning) or after charging from flat(ish) and not every time either, I just wish I could get to a certain set of circumstances where I could say if X happens it resets but I can’t, in other words it’s unpredictable hence why if it were me I think I’d give up and see if long term the cause get’s revealed.

 

I’m rather hoping that this morning was fluke as if that doesn’t repeat then we should be good, the only real “error” of any kind I have been able to find was the original RTC battery only showing a little over 3 volts whereas the new one of course is 3.3.

 

So, do we close it or leave it open? The other problem of course being it takes ages to test anything worthwhile out.

 

Phil

projectkk2glider, 2015-08-05:
Leave it open.

kilrah, 2015-08-05:
I'd vote for a hardware issue... would be worth looking at the supply rail when powering up. If it rises too slowly it could cause that kind of things from what I remember reading in the datasheet long ago.

hrenz, 2015-08-06:
Hy, 

what type is the RTC Battery:  2032  or other type,  one or two

myckey, 2015-08-06:
It's a CR1220 and that is what I asked for and got when I bought the replacement fo it.

 

Version: 2015.0.6086 / Virus Database: 4392/10383 - Release Date: 08/06/15

hrenz, 2015-08-06:
Hy,
for the RTC its a CR 1220  Lithium-Knopfzelle, 3 Volt, 35 mAh, 12,5x2,0mm

bsongis, 2015-08-06:
It's an hardware problem, please contact FrSky (the 100uF is not enough, they will use a 470uF instead)

closed 2015-08-06 by bsongis

Spainman, 2015-08-07:
Bertrand, would you mind pointing out where the corresponding cap is located so that we can just change it instead of going through the hassle of asking frsky for help? Is it the supplementary non-smd one near the rotary encoder?

projectkk2glider, 2015-08-07:
> Is it the supplementary non-smd one near the rotary encoder?

**First a disclaimer**. I do not have the new board in front of me! I am giving this advice based on photos from rcfourms and on knowledge of previous version of X9E board.

The C71 SMD capacitor is the original one on the 3.3V output from the regulator. Its value is 22uF. The additional non-smd capacitor looks to be in parallel with it. If that is the case and its value is 100uF then this is the cap that was added later to help with the RTC issue. 

Now it was determined by FrSky that 100uF is not enough and should be replaced with a 470uF.

But the 100uF cap looks to be quite inaccessible, it is blocked by the small board for the rotary encoder. So, my suggestion is:
 * leave that cap alone
 * add additional capacitor in parallel somewhere betwen Vcc and GND lines, the best place so far I have found (again from the photos) is on the right side of the board by the Bluetooth module.

Take a look a this photo http://static.rcgroups.net/forums/attachments/5/5/9/3/9/6/a8081298-73-DSC_8543.jpg

You will see a flat cable connector named P4 on the left from the Bluetooth module. The pin 1 on P4 connector is on the bottom and is labeled 1GND. The pin 4 on that connector is the Vcc line we are after. If you follow the trace (it is quite wide) you will se it going to the Bluetooth module (second connection from the top on the left side of the module) an also to the bottom and left (near the P4 letters) an it ends with two vias.

That trace is your Vcc line and should have 3.3V when the radio is powered on. The GND is on the copper fill surrounding it.

Now if you have enugh space you can put 470uF capacitor (at least 6V rating):
 * on the left top-most pins on the Bluetooth module. The top one is GND (minus) the second one (below it) is Vcc (plus).
 * another alternative is to scrape of the protective layer from the board near the P4 sign, both the Vcc trace and the GND plane beside it and solder it there. That is what I reccomend, but be careful you don't brake any existing trace or make a short circuit.

**And do all of this with the main battery disconnected.**

If you decide to experiment yourself, try it first with a 470uF cap, if that does not help try increasing it to 1000uF and report back on  your findings. Good luck.

Spainman, 2015-08-07:
Thanks a lot for taking your time! Great instructions!

I just had a look at the board and it has two ceramic smd caps (c70, c71) plus the non-smd elko, all in parallel. 

Indeed there is no access without removing the encoder board and it might be easier for the general do-it-your-self-er to just add a cap as you described. I have some experience and the required equipment so I would prefer to simply replace all three caps with one 1000uF smd elko (10x10mm) right on the board in place of c70/71 and keep thing kind of 'clean'.  It should fit mounted across the pads of c70/71 diagonally

Could that represent a problem due to the different characteristics of the elko and the original ceramic caps which I would omit? I really have no clue when it comes to analog stuff, so do you consider the original ceramic caps mandatory? Thank you.

projectkk2glider, 2015-08-07:
>Could that represent a problem due to the different characteristics of the elko and the original ceramic caps which I would omit? I really have no clue when it comes to analog stuff, so do you consider the original ceramic caps mandatory?

Leave original caps C70 and C71 on the board! They are necessary for the switching regulator. They have different characteristics (lower ESR). You could remove existing 100uF electrolytic capacitor and replace it with a bigger one.

hrenz, 2015-08-07:
think there is the wrong picture above
its shows only the BT side

projectkk2glider, 2015-08-07:
@hrenz correct, the caps C70/71 are on the left side. The picture above is for my suggestion at the placement of the additional cap.

hrenz, 2015-08-07:
do you have detail picture of the left side with C70/71

projectkk2glider, 2015-08-07:
Pictures I used I found in this thread http://www.rcgroups.com/forums/thumbgallery.php?t=2398179&do=threadgallery

Spainman, 2015-08-07:
Great, thank you. I will replace the add
on elko then and report back after a few days of testing.

Here is a detail picture for reference.
[image]
-

myckey, 2015-08-07:
Good luck with the testing but just for information in the last four days or so I've been unable to get mine to reset its RTC no matter what I do to it.

 

 <https://cloud.githubusercontent.com/assets/8582669/9144713/fc7c7dbe-3d4e-11e5-88fe-05116703e076.jpg> img_20150807_215256 1

Version: 2015.0.6086 / Virus Database: 4392/10390 - Release Date: 08/07/15

projectkk2glider, 2015-08-08:
According to FrSky the 100uF capacitor is borderline, so it works on some (most) radios and fails on others (hopefully the minority).

MikeBland, 2015-08-08:
I remember we had some problems during development with the RTC. When shutting down, I have this code in use for the STM processor:
				soft_power_off() ;		// Only turn power off if necessary
#ifdef PCBX9D
				for(;;)
				{
					wdt_reset() ;
  				PWR->CR |= PWR_CR_CWUF;
  				/* Select STANDBY mode */
  				PWR->CR |= PWR_CR_PDDS;
  				/* Set SLEEPDEEP bit of Cortex System Control Register */
  				SCB->SCR |= SCB_SCR_SLEEPDEEP_Msk;
  				/* Request Wait For Event */
  				__WFE();
				}
#else

It might sort the RTC problem.

Mike.

hrenz, 2015-08-08:
Hy,
have changed from 100uf/63V to 470uF/25V  at same place
no Problem
[image]
cut the wire of the original elko , sold at the wires the new one

Screws of the Battery holder are to short took longer from servo holders

projectkk2glider, 2015-08-15:
@Spainman and @hrenz how is your RTC doing? Any better?

>What about P12 4pins?

The P12 is the serial port, same as in Taranis, https://github.com/opentx/opentx/wiki/Taranis-serial-port

>4 pin row left under the batteryholder for?

That is P8 and its a SWD debug port for STM32 processor. Not useful for other purposes although it has a GND (pin 4) and VCC (pin 1) signals.

Spainman, 2015-08-15:
Actually I did not experience the problem on my x9e but I ebayed it after a few days without really using it so I don't know if it was affected. The transmitter I modified to a 1000uF cap belongs to a friend and as far as I know the reset did not happen since. But most likely it wasn't used frequently enough to come to any conclusions. As far I know it's going to change hands too so I'm afraid I won't be able to report anything useful... sorry

projectkk2glider, 2015-08-21:
Good news, post on forums http://www.rcgroups.com/forums/showpost.php?p=32476138&postcount=763:

>First, I replaced the battery because voltage was only 2.8 V. Didn't change anything, there was still a reset from time to time. Then I replaced the 100uF capacitor with a 220uF. Now it works fine, no resets any more.