opentx/opentx #2627: Telemetry - 40 Amp Sensor - Vfas & Curr - Speach Wrong

rguptaca, 2015-08-03:
open-tx-Taranis-2.1.0
2015-07-23
08:48:31
217
Taranis (not plus)

See video link below.
Telemetry shows correct values.
Speech:
- wrong vanues
- sometimes speaks up to 1 decimal place and sometimes no decimal places. Precision set to 1 & 2 (testing) makes no difference
- many units such as W (watts) speech says something totally different such as "hours". 

Could not find Cnnsp (which I was using).
Have not been able to locate explanation for:
- Vfas+ Vfas-, Curr+, Curr-
- Auto offset, Positive, Persistent (Telemetry  sensor screen)

rguptaca, 2015-08-03:
Video 1
https://youtu.be/bePReQWWgnw
Video 2
https://youtu.be/WucWM5jPlJY

lichtl, 2015-08-03:
Hi,

did you replaced/uploaded the new sound files? With 2.1 there was some changes so you need to update the folder on your SD card

closed 2015-08-03 by kilrah

rguptaca, 2015-08-05:
That fixed it.