opentx/opentx #2630: S.Port High Precision Vario not giving correct initial value

bitogre, 2015-08-04:
I use the High Precision Vario with a X6R receiver connected via S.Port.  While this seemed to create the sensor fine, the new OpenTX 2.1.0 on a Taranis it is reporting that my current altitude is -116ft when on OpenTX 2.0.x it would indicate that my model was a 0ft from where telemetry was last reset.

griebd, 2015-08-04:
Have you set "auto offset" for that particular sensor?

closed 2015-08-04 by projectkk2glider

bitogre, 2015-08-04:
griebd, what do you mean by "auto offset"?  I am using the values that the Taranis automatically supplied for the sensor.

projectkk2glider, I see no indication that this was fixed, is a duplicate, or as designed so why did you close this bug?  If it is fixed, I'd like to know how.  If it is duplicate, what is the issue number that it is a duplicate of?  if it is as designed, please explain why this behavior was changed.

kilrah, 2015-08-04:
It was closed because it is as designed and you got the answer, you need to check Auto Offset in the sensor parameters.

http://openrcforums.com/forum/viewtopic.php?f=45&t=6887

bitogre, 2015-08-04:
kilrah, thanks for the additional information.  It is useful and needed to address the issue and should have been provided in griebd's or projectkk2glider's comments.

However, I would argue that the bug should not have been closed until I had a chance to respond to the question posed by griebd.  What if it was set, I had properly reset that sensor, and it was still showing the -116ft?  I would think that indicates that it is not functioning as expected (I nee to get home to try this out to see).  Maybe if I took over a week to respond, then it might be reasonable to close without knowing if "auto offset" was set.

Also, I think this bug should have remained open as now I think the bug it that "auto offset" is not set by default when the sensor is created.  I think this should be the default to keep the behavior the same as it was on OpenTX 2.0.x.

kilrah, 2015-08-04:
> What if it was set, I had properly reset that sensor, and it was still showing the -116ft? 

Then we'd have reopened the issue... but we know there is 99% chance that the problem was Auto Offset not being checked, so closing to remove clutter. You're not the first one to miss that, and knowing 2.1.0 has been out for 3 weeks if there was really a bug in something used as commonly as that we'd probably already know it.

The decision to not check auto offset by default is deliberate so as not to alter the received value by default. The "wrong" behavior was the one in 2.0 but there wasn't really another option there.

bitogre, 2015-08-04:
I guess that is reasonable but it can leave new users with a bad experience and potentially a negative impression of the community that is working on the firmware.  Is it really that hard to provide more detail before closing a bug?  Or is it really that problematic to leave the bug open a few extra days so these new users can feel like they are not being ignored?

BTW, while the link was helpful, it does a poor job of explaining what "auto offset" does.  And I am surprised the fact the altitude was negative did not raise any concerns (I do not live some place that is below sea level).

kilrah, 2015-08-04:
> Is it really that hard to provide more detail before closing a bug? 

When a very small team of developers closes 5-10 non-bugs a day they do tend to become... "brief and efficient" ;)
If you have usage questions or are not sure about something you'll have more success, and should first document yourself or seek help on user forums where you'll have a much higher number of fellow users to help you. Ideally this place should be exclusively used for reporting confirmed issues you're sure can't be related to misunderstanding or usage errors.

http://openrcforums.com/forum/viewtopic.php?f=45&t=7239

As explained there 2.1 is very new and kind of "experimental", things may still change before it becomes "stable", and it will likely take a few months for documentation to be widely available (as there now is for 2.0).
If you want to use it now expect to do some research of your own and to have to figure things out by yourself.

kilrah, 2015-08-04:
Ah and about the negative alt... we just display what the sensor sends. It's a barometric sensor which would need to be calibrated to the day's atmospheric conditions, as there isn't a way to do it it's normal for reported altitude to be significantly off in either direction from one day to another.

bitogre, 2015-08-04:
That is also good to know but it is very hard to find that forum.  I suspect most people find info about this from http://www.open-tx.org/ which does not mention the forum on the main page.  On the links page, it describes the forum as "developer forum" so users are unlikely to go there.  More prevalent on links page is OpenTX Users Mailing List but that looks like it is barely used.  Finding the place to report issues was a lot easier than finding the user forums.  Maybe a link should be added to the main page of http://www.open-tx.org/ to the forum so it is easier to find.

I have no problems using "experimental" code so long as I know it is that.  But, until I got into this discussion with you, I had no indication that 2.1 is currently "experimental".  No where on the [News announcing its release](http://www.open-tx.org/2015/07/16/opentx-2.1.0/) does it indicate it is "experimental" or not ready for most users.  Even [your post on the forum](http://openrcforums.com/forum/viewtopic.php?f=45&t=7239) is not clear about it being experimental.  I had to carefully read between the lines to find where you give a very subtle implication that 2.1 was not stable by referring 2.0 the stable release.  Even on the [links page](http://www.open-tx.org/links.html), the is a link under "OpenTX Nightly Builds" to "2.1 - master branch (**stable**)" implies that 2.1 is not "experimental".

I am sorry for dragging this out.  I do not want to be difficult.  I want to help and I feel that others are likely to run into the same problem if I do not point out how I got the wrong impression.

kilrah, 2015-08-04:
No problem, thanks for following up. We're trying to improve that side of things, but... we're primarily coders and the "boring" doc side is often left behind, especially since 3rd-parties have started taking care of that (open-txu.org for example).

The link to the thread on openrcforums.com has been added to the release page.

Ah and by the way, [Downloads](http://www.open-tx.org/downloads.html) page clearly (I think) states:

> As we are dependent on user feedback, a major release will start as “unstable” and will go through several rounds of refinement based on the gathered user experiences, which can initially cause radical changes in the way a feature operates between minor versions. Once things have settled and we have decent documentation available that major version will become “stable” i.e. its features will be frozen, and from that moment on any new minor versions would only be released to fix bugs that may be discovered.

For example 2.0 could be considered stable after about 12 revisions.