opentx/opentx #2631: Incorrect limit in Logic Switch for High Precision Vario

bitogre, 2015-08-04:
I use the High Precision Vario with a X6R receiver connected via S.Port.  When trying to set up a logic switch and special function to report my altitude anytime my aircraft exceeds 350ft but it seems like Taranis OpenTX 2.1.0 limits the altitude to 300ft.  I was able to set this limit as desired on Taranis OpenTX 2.0.x.  This maybe related to Issue #2630 with the automatic sensor creation did not create the sensor correctly.

kilrah, 2015-08-04:
Are you setting the limit on the radio or in companion?

bitogre, 2015-08-04:
When I first reported it, I was only looking at Companion.  However, I did a test this morning and determined the issue exists on both.  It behaves the same way on both the Taranis and in Companion.

projectkk2glider, 2015-08-05:
Confirmed on simu and in Companion simulator. When the precision of Alt is 2 the max value is 300.00m. When the precision is 1, the max value is 3000.0m. I guess this is intended! So @bitogre set precision to a lesser value. Setting precision to 0 will allow up to 30000 m (of ft).

Some screenshots, PREC2 on Alt:
[image]

PREC1 on Alt:
[image]

fraternl, 2015-08-05:
Why would anyone set the precision of the altimeter to centimetred?

projectkk2glider, 2015-08-05:
Well, it is the same for all other telemetry values where higher precision makes more sense. There is no special case for altitude. I think it is OK.

bitogre, 2015-08-05:
I agree with @fraternl, it does not make sense for the precision of the altimeter to be cm.  This was auto created by the firmware with the highest precision possible.  Looking at the code (specifically line 70 of [radio/src/telemetry/frsky_sport.cpp](https://github.com/opentx/opentx/blob/master/radio/src/telemetry/frsky_sport.cpp)), it should be easy to change the default precision of the altimeter from 2 to 1 or even 0.  Had this default been lower, it would have been more user friendly and I would not have had this issue in the first place.

I think the same would apply to Air and GPS speeds and GPS Altitude.  Precision is nice to have but is not useful or even valid, especially when the values are going to be large and the measuring device has limited accuracy.  Very rarely is more than 3 digit useful so if a value is expected to exceed 100 (which is likely for Airspeed and Altitude), having a precision greater than 0 is very questionable, especially of the measuring device has an error greater than 1% (which is common).

BTW, the [specs on the High Precision Variometer sensor](http://www.getfpv.com/radios/radio-accessories/frsky-variometer-sensor-w-smart-port-high-precision-version.html) say it is only good to 0.1m so having the default of 2 is a default higher than the device is capable of.  Based on this specified precision, the default should be 1.

projectkk2glider, 2015-08-05:
I agree, lets change the default precision for some sensors.

commit 2378080a (2015-08-05): Fixes #2631
closed 2015-08-05 by bsongis