opentx/opentx #2633: Vector and X9D not working with 2.1.0

MRTJ30, 2015-08-04:
Hi guys I have upgraded my Tarnish plus to latest firmware 2.1.0 for X9D. It displays a message box All Telemetry boxes are full and press exit but it doesn't exit the screen stays there. Please fix this. I am trying to reflash back to 2.0.9

kilrah, 2015-08-04:
Eagletree needs to fix their software, they're aware of the issue. 2.1.1 will have an interim solution.

If you reflash to 2.0.9 you'll need to reload an eeprom backup done before the upgrade or you'll lose all your settings.

closed 2015-08-04 by kilrah

MRTJ30, 2015-08-04:
just done it thank you. hope they fix it soon. also remembered I should have checked the FrSky site first. They say if it aint broke don't fix it. all the new screens looked so sexy hope they release update soon