opentx/opentx #2634: Remove the "download nightlies" checkboxes in companion

kilrah, 2015-08-04:
Or at least have a build option so they only appear on a version of companion that is already a nightly build. 

These past days I've seen several people on various forums telling general users who have no idea of what they're doing to switch to nightlies without warnings or anything.

fraternl, 2015-08-04:
Then let them get punished for it   ;-)

kilrah, 2015-08-04:
But the person who might be "punished" is not the one who should be. The problem is the people who tell normal users to go and use nightlies, not the users themselves who wouldn't do it on their own.

I've now seen 4 times in the past couple of days people posting about a problem in 2.1.0 that is now solved, and someone just telling them "use the nightlies, it's fixed" without further warning. As it's just a box to tick it doesn't seem like a big deal, and unfortunately there's no way we can force these people to say "wait for the next release, I've tested the nightlies and can confirm it will be fixed"... so we have to make them a bit less accessible or at least add very visible warnings.

fraternl, 2015-08-04:
I was only joking....
IMHO as things are now I would go and release current state as 2.1.1 and then 2.1.2 on top...
It's not up to me and you probably have more than valid reasons not to do so...
The only thing is that current 2.1.0 has too many obvious bugs to make people upgrade from 2.0.x

Hence there are people who encourage others to use the nightlies.
I'm for sure guilty of that, but AFAIK not publicly (some close friends).
But I do tell them to go on normal releases when 2.1.1 comes out.

They only bought their first Taranis last month and I didn't want to explain stuff 2 times...

kilrah, 2015-08-04:
2.1.1 definitely is overdue, but the balance between requiring users to do multiple updates and making sure to get most of the big serious bugs out in one go is hard to find. 
And in the end we did tell users that they should only jump right on 2.1.0 if they were somewhat adventurous, because a .0 release is bound to be a bit sketchy - so now they're punished for coming too early ;) 
We'd like 2.1.1 not to be sketchy and to be able to tell people they can go to it without having to expect one release per week.

griebd, 2015-08-04:
I agree that it's already a good time for 2.1.1!
Just put on the update notes the critical known bugs (the 2 issues still open).
By releasing 2.1.1 in the actual state there will be a lot less bugs opened every day!

By the way, I think there should be a more verbose announcement about updating the sound pack for those upgrading from 2.0. (many bugs were opened where the cause was failure to upgrade the sound pack)

And back to this issue, I too vote for NOT having the option to download the nightlies on the release version. I know, freedom. But then, the radio is getting more and more popular, people are using it just following someone else guidance. The more dangerous options should be a bit more hard to find (but still available somehow).

vadav, 2015-08-04:
A clear warning next the the box would be fine.

kilrah, 2015-08-04:
Not easy to put a 100 word explanation there...

fraternl, 2015-08-05:
"Do NOT change!"

commit a8ff8b52 (2015-08-05): Fixes #2634 - Option ALLOW_NIGHTLY_BUILDS added
closed 2015-08-05 by bsongis