opentx/opentx #2638: FrSky 9XE and companion 2.10

myckey, 2015-08-04:
Using latest versions of both companion and firmware as downloaded by companion.

Companion cannot open the log files the TX has created, no problem opening in Excel and the data looks good and "clean", error message in Companion is "selected file contains X invalid lines out of X total lines". Same result if tge log file is transferred to desktop first then opened.

kilrah, 2015-08-04:
What sensors do you have?

myckey, 2015-08-04:
No sensors as such just a D8R 11 Plus RX.

 

No virus found in this message.
Checked by AVG - www.avg.com <http://www.avg.com> 
Version: 2015.0.6086 / Virus Database: 4401/10370 - Release Date: 08/04/15

kilrah, 2015-08-04:
Can you post an affecting log? Note the email reply doesn't support attachments.

myckey, 2015-08-04:
Here is a small log file opened in Excel, no RSSI as the RX wasn't on but all error messages are the same.

 

Version: 2015.0.6086 / Virus Database: 4392/10370 - Release Date: 08/04/15

myckey, 2015-08-04:
[image]

myckey, 2015-08-04:
Log file now displayed.

kilrah, 2015-08-04:
Would be best to have the original file so I can try to reproduce.

myckey, 2015-08-04:
Tell me how to send it and I will do so right now

kilrah, 2015-08-04:
You can use something like wetransfer.com and post the link.

myckey, 2015-08-04:
I've put two log files on dropbox for you, https://www.dropbox.com/sh/zpxakqj5agwl06g/AACttaabxnnvKxHtFs1G-7Yga?dl=0

projectkk2glider, 2015-08-04:
@myckey your log file has three more values in data rows compared to the first row where field names are declared. This is the first two lines of your log:
```
Date,Time,RSSI,Rud,Ele,Thr,Ail,S1,S2,S3,LS,RS,SA,SB,SC,SD,SE,SF,SG,SH
2015-08-04,11:35:15.360,105,-3,-2,-1024,0,0,12,-1024,0,-13,17,1024,1024,-1,-1,-1,-1,-1,1,-1,-1
```

This has either been already fixed in 2.1.1 **or** does not occur if one logs some additional telemetry fields, because it works for me (with several telemetry fields enabled). I haven't tried with default fields only like your example.

griebd, 2015-08-04:
I have a model without telemetry (DSM2) and the logs worked as expected, BUT I'm on a self compiled 2.1.X...

Actually, now that I think about it, I did not configure this model to log anything at all... Why are logs been generated? I though logs where enabled only by setting a 'special function'...

kilrah, 2015-08-04:
What radio do you have? If it's an X9E it could have something to do with the number of switches as it's variable.

kilrah, 2015-08-04:
Ah sorry it IS an X9E, was in the original title.

myckey, 2015-08-04:
I can't say it would log or wouldn't if you hadn't enabled it but I enabled logs in the special function menu, files as said can be read in companion OK if the last three columns of non headed data are deleted.

Question, should I be raising an issue here if I'm only using the latest download from Companion? I assume 2.1.1 isn't on general release yet which is why companion doesn't update to it?

kilrah, 2015-08-04:
Correct, 2.1.1 isn't released yet. Anyway it's a firmware issue, not a companion one.

PR #2653 (2015-08-06): Fixes #2638: header line for logs fixed for 9XE
Closes #2638

Not pretty but it works (tested in simu).
commit 20eff335 (2015-08-06): Fixes #2638: header line for logs fixed for 9XE
commit 6fcecb56 (2015-08-06): Merge pull request #2653 from opentx/projectkk2glider/issue_2638_9xe_logs

Fixes #2638: header line for logs fixed for 9XE
closed 2015-08-06 by bsongis