opentx/opentx #2639: X9E Faders S1,S2 - Crosstalk and Deadband

Spainman, 2015-08-04:
Two observations:
1) Moving the left fader slightly alters the reading of the right fader by up to 1% (reading in channel monitor, mechanical crosstalk ruled out by removal of the board and testing outside the case)

2) The faders reach their maximum / minimum readings at round about 80% of their available mechanical travel , calibration doesn't help (tested with the board mounted in the case to rule out over travel due to possibly missing mechanical stops)

Is this fixable in firmware or are these HW issues? 
Thanks in advance!

projectkk2glider, 2015-08-05:
Point 1. Should be confirmed that it happens on other radios not just yours. But it can be explained why it happens and also fixed (improved) in firmware. I guess the slider have a higher impedance compared to other analog inputs and thus require longer ADC acquisition time. But as I wrote above, mention this issue on forums and try to get the same behaviour confirmed from other users. Then we will look at this issue more deeply.

Point 2. is a mechanical issue. The sliders used have a large portion of slide travel at both ends where the resistance does not change. I noticed that when we first received testing boards and pointed that out to FrSky. I hoped that they would either change the slider or limit the travel mechanically with the case opening. Looks like they did not. Can't think of anything that can be done in firmware to solve this.

fraternl, 2015-08-05:
I haven't noticed either. Maybe I shouldn't react until I investigated. Will do soon.
If "1" is true I would expect it to be electronic crosstalk during the A/D conversion. 

I thought I had the opposite of "2" and it never reached a maximum
I therefore made a curve that would create a deadband at the borders.

Spainman, 2015-08-07:
Thank you for looking into it. 
Considering that point 2 is not fixable and that my right fader now started to jam when actuated with the slightest lateral force, I will try to find quality components that don't have any of these issues.  This might change the crosstalk to the better or worse so I think that until somebody else brings up point 1, there is no reason in taking any actions particularly for S1 and S2. 

Nevertheless I have a __suggestion for an enhancement:__ 
It might be worth considering a parameter in the hardware configuration screen to allow for a individual customisation of the ADC acquisition time per pot. My thought here is that the x9e has free analog inputs that might be populated with arbitrary pots, faders, joysticks or even a force sensitive resistance (great for brakes) that might produce similar crosstalk due to the 'impedance thing'.

janekx, 2015-08-08:
1 confirmed, if look to annalog screen left slier influence right 10: 415-41A

projectkk2glider, 2015-08-08:
Thank you both for the tests, we will look into it.

projectkk2glider, 2015-09-20:
@bsongis just a friendly reminder, you have the production X9E...

janekx, 2015-09-20:
Will test it too have some on stock

bsongis, 2015-09-20:
No I don't have the production radio, just a more recent prototype than yours!

janekx, 2015-09-20:
Ok just tested on production line.  Right slider not influenace left but left influence right.  
The analog goes from 0411 to 0415 when left goes from - 100 to 100

bsongis, 2015-10-13:
Is there anything we can do in firmware for this one? I delay to 2.1.5, but feel free to close it!

projectkk2glider, 2015-10-13:
We could try to have a longer ADC conversion time, which would solve the problem if the problems is too large input impedance. But for that to happen, one of the devs has to have the production hardware for testing.

janekx, 2015-10-25:
You can compile me the build to test

projectkk2glider, 2015-10-25:
@janekx we did nothing to affect this issue since it has been reported, so any custom build to test/debug this issue will first have to be developed.

projectkk2glider, 2015-11-20:
I have X9E now and I confirm the issue. Only S1 influences S2, the change is 5-7 out of 0x7FD (2045) which is about 0.24%. Is this enough to warrant a change in firmware?

janekx, 2015-11-20:
We have discussed the channels in Taranis and their oscilation, the oscialting about 0.8% is in specs...

closed 2016-01-18 by projectkk2glider

projectkk2glider, 2016-01-18:
For reference, as well as setting sample time on proper ADC channels, I had to increase sample time one step. With the original sample time there was still some (albeit smaller) cross-talk present. 

Increased sample time should not affect the time it take to do the ADC reads, since ADC1 has more channels to sample.

commit a7924216 (2016-01-18): Fixes #2639: ADC crosstalk on X9E fixed, sample time was not set for the right channels, sample time also needed to be increased
commit 46bdd0eb (2016-01-18): Fixes #2639: ADC crosstalk on X9E fixed, sample time was not set for the right channels, sample time also needed to be increased
commit a8172e35 (2016-01-18): Fixes #2639: ADC crosstalk on X9E fixed, sample time was not set for the right channels, sample time also needed to be increased