opentx/opentx #2640: Special functions: saying telemetry value istn't available since 2.1

dklueh79, 2015-08-04:
Until today I used the chance to set a special function for "say value" for telemetry data.
But since upgrading on 2.1, it isn't available any more.

Could this be readded, please?

kilrah, 2015-08-04:
It's of course still there. You need to let the radio discover the connected sensors before they can be selected as value to say.

closed 2015-08-04 by kilrah

dklueh79, 2015-08-06:
OK, that works nearly.- But I didn't find any documentation, if the old sensors still work (analog, with sensor hub on an D8R-Receiver)
During switching on the Taranis the old FAS-100 and FLVS-01 aren't shown...but one of the old Temp.Sensor. The screenshots show the new and the old telemetry window.

[image]
[image]

kilrah, 2015-08-06:
That capture shows all the hub data, but it seems your fas/flvs weren't connected.

dklueh79, 2015-08-06:
OK, but I haven't cconnected fuel- or apm-sensor- why does the taranis recognize this?

So, the old sensor all should work, too- right?

kilrah, 2015-08-06:
That's just how the hub works... It sends values for everything even if the sensor is not connected... Except for the fas/flvs that use a slightly different interface.

All sensors are of course supposed to work.