opentx/opentx #2643: Moving lines in mixer leave stuck on the top, not moving down

janekx, 2015-08-05:
Tested on opentx-taranisx9e-lua-mixersmon-cz-2.1.1N404
Taranis-E
Reproduction
goto mixer 
go to any line of mixer and pres rotary 
rotate and the line will move down the choosen line 
but after move down will move only up to top and then will be stucked there and not possible to move down.

projectkk2glider, 2015-08-05:
Did it happen with older builds also? Like before the 70e93fb4935ab660562446bdde37451f8ee41f83 was introduced, that would be Build `#399` or before.

janekx, 2015-08-05:
Tested on N399 and N404 to be sure still not fixed. 
At N399 the rotary not work for new model lua script, on N404 work but this still not.

bsongis, 2015-08-05:
Yes the problem comes from the ROTARY events in menu_model_input_mixes.cpp
I just fixed it, and I will push the commit in one hour

janekx, 2015-08-05:
Will check it in upcomming nightly ....

bsongis, 2015-08-05:
Give me a few minutes, I will manually launch a nightly build. I would like to release today ... Still 3 hours left

closed 2015-08-05 by bsongis

bsongis, 2015-08-05:
It should be ok now:
http://downloads-21.open-tx.org/nightly/companion/
The N405

bsongis, 2015-08-05:
OOps I forgot to enable the new Companion option (for nightly access)

janekx, 2015-08-05:
Now it seems work OK thanks
opentx-taranisx9e-lua-mixersmon-cz-2.1.1N406

commit 15469aea (2015-08-05): Fixes #2643