opentx/opentx #2644: Gspd value in Taranis Telemetry too high

marcodcribeiro, 2015-08-05:
hi,

i have taranis 2.1.0 with telemtry but gspd value in display show values like 400 and so.
any update on that?

projectkk2glider, 2015-08-06:
Sport or hub sensor?

marcodcribeiro, 2015-08-06:
its a external GPS and the Telemetry is by Naze32.
the unit i try kmh an kts.
the value in naze32 logs its correct

kilrah, 2015-08-06:
What receiver? What firmware and version on the Naze32?

The problem is most likely in the Naze32 firmware as we've had no reports of issues with the FrSky GPS.

marcodcribeiro, 2015-08-06:
Sorry,

i'm use Naze32 Cleanflight 1.9.0 and frsky d4r II

projectkk2glider, 2015-08-06:
Bug in Cleanflight https://github.com/cleanflight/cleanflight/blob/master/src/main/telemetry/frsky.c#L248

The sent gps speed is multipled with 0.1944 instead of 0.01944

kilrah, 2015-08-06:
I'll send a pull request to cleanflight, already have a fork

closed 2015-08-06 by kilrah

marcodcribeiro, 2015-08-06:
Tks for the help,
So or change to baseflight or wait to any update to cleanflight.

kilrah, 2015-08-06:
Baseflight doesn't send GPS speed at all.

marcodcribeiro, 2015-08-06:
Tks,
so for now i'll create a LUA script to do the calculation about GPS

projectkk2glider, 2015-08-06:
You could also play with the ratio parameter.

marcodcribeiro, 2015-08-06:
how?

projectkk2glider, 2015-08-06:
Ah, I was wrong. The Ratio does not support division by 10. You will have to wait for the #2615

kilrah, 2015-08-06:
It will once #2615 is implemented :\

marcodcribeiro, 2015-08-06:
Tks guys for the help. I'll wait

hydra, 2015-08-07:
merged https://github.com/cleanflight/cleanflight/pull/1191 into cleanflight