opentx/opentx #2645: Companion 2.1.1N407

rguptaca, 2015-08-06:
I uninstalled both 2.0 and 2.1 using the uninstall program provided.
Downloaded 2.1.1N407
It wanted to create/install into 2.0
See pic
[image]

projectkk2glider, 2015-08-06:
I saw the same issue. It seem that installer program somehow (I bet it is in the registry) remembers the last installation path for Companion and offers it in this dialog despite the fact that default values in cmake are configured differently.

kilrah, 2015-08-06:
Yes and it's desired so that if the user chooses another directory it's remembered on subsequent updates. The problem is when settings are imported from a previous version, this particular key should be skipped.

projectkk2glider, 2015-08-06:
When installing on a fresh Windows machine, first installed 2.0.17 and then 2.1.1 nightly `#407` I don't have this problem. It seems that some other version of Companion messed up that setting in the registry. 

@rguptaca what do you have in the registry key `HKEY_CURRENT_USER\Software\OpenTX\Companion 2.1`?

kilrah, 2015-08-06:
It might only happen if the user did not go through 2.0.17 and updated straight from <2.0.17 to 2.1.

2.0.17 moved the settings to the "2.0" key to allow for concurrent versions, before that it wasn't expected to have multiple versions.

projectkk2glider, 2015-08-06:
@kilrah I understand that, but I can't see where the 2.1 is picking up the directory of 2.0. It is looking in completely different registry key which should not even exist!?

kilrah, 2015-08-06:
Will have to look again in detail

rguptaca, 2015-08-07:
Attached are the Registry entries for Comopanion 2.1 

Key Name:          HKEY_CURRENT_USER\Software\OpenTX\Companion 2.1
Class Name:        <NO CLASS>
Last Write Time:   2015-08-06 - 1:31 AM
Value 0
  Name:            <NO NAME>
  Type:            REG_SZ
  Data:            C:\Program Files (x86)\OpenTX\Companion 2.1

Value 1
  Name:            Start Menu Folder
  Type:            REG_SZ
  Data:            OpenTX Companion 2.1

Value 2
  Name:            arm_mcu
  Type:            REG_SZ
  Data:            at91sam3s4-9x

Value 3
  Name:            avr_arguments
  Type:            REG_SZ
  Data:            

Value 4
  Name:            avr_port
  Type:            REG_SZ
  Data:            

Value 5
  Name:            avrdudeLocation
  Type:            REG_SZ
  Data:            

Value 6
  Name:            backLight
  Type:            REG_DWORD
  Data:            0

Value 7
  Name:            backupEnable
  Type:            REG_SZ
  Data:            true

Value 8
  Name:            backupOnFlash
  Type:            REG_SZ
  Data:            false

Value 9
  Name:            backupPath
  Type:            REG_SZ
  Data:            C:\Users\Ravi\Documents\OpenTX\Backup

Value 10
  Name:            checkHardwareCompatibility
  Type:            REG_SZ
  Data:            true

Value 11
  Name:            dfu_arguments
  Type:            REG_SZ
  Data:            -a 0

Value 12
  Name:            dfu_location
  Type:            REG_SZ
  Data:            

[... 439 lines cut ...]

Value 22
  Name:            PPM_Multiplier
  Type:            REG_DWORD
  Data:            0

Value 23
  Name:            VbatCalib
  Type:            REG_DWORD
  Data:            0

Value 24
  Name:            vBatWarn
  Type:            REG_DWORD
  Data:            0x41

Value 25
  Name:            VbatMin
  Type:            REG_DWORD
  Data:            0xffffffe2

Value 26
  Name:            VbatMax
  Type:            REG_DWORD
  Data:            0xffffffd8

rguptaca, 2015-08-07:
Attached are the Registry entries for Comopanion 2.0
I removed 2.0 using the uninstall script provided so not sure why these are still here.

Key Name:          HKEY_CURRENT_USER\Software\OpenTX\Companion 2.0
Class Name:        <NO CLASS>
Last Write Time:   2015-06-04 - 6:31 PM
Value 0
  Name:            <NO NAME>
  Type:            REG_SZ
  Data:            C:\Program Files (x86)\OpenTX\Companion 2.0

Value 1
  Name:            Start Menu Folder
  Type:            REG_SZ
  Data:            OpenTX Companion 2.0

Value 2
  Name:            arm_mcu
  Type:            REG_SZ
  Data:            at91sam3s4-9x

Value 3
  Name:            avr_arguments
  Type:            REG_SZ
  Data:            

Value 4
  Name:            avr_port
  Type:            REG_SZ
  Data:            

Value 5
  Name:            avrdudeLocation
  Type:            REG_SZ
  Data:            

Value 6
  Name:            backLight
  Type:            REG_DWORD
  Data:            0

Value 7
  Name:            backupEnable
  Type:            REG_SZ
  Data:            true

Value 8
  Name:            backupOnFlash
  Type:            REG_SZ
  Data:            false

Value 9
  Name:            backupPath
  Type:            REG_SZ
  Data:            C:\Users\Ravi\OneDrive\Documents\Taranis

Value 10
  Name:            dfu_arguments
  Type:            REG_SZ
  Data:            -a 0

Value 11
  Name:            dfu_location
  Type:            REG_SZ
  Data:            

Value 12
  Name:            embedded_splashes
  Type:            REG_DWORD
  Data:            0x1
[... 353 lines cut ...]

Value 21
  Name:            PPM_Multiplier
  Type:            REG_DWORD
  Data:            0

Value 22
  Name:            VbatCalib
  Type:            REG_DWORD
  Data:            0

Value 23
  Name:            VbatMin
  Type:            REG_DWORD
  Data:            0

Value 24
  Name:            VbatMax
  Type:            REG_DWORD
  Data:            0

Value 25
  Name:            vBatWarn
  Type:            REG_DWORD
  Data:            0

rguptaca, 2015-08-07:
Installed 2.1.1 public release. Works now. I guess it got fixed.

kilrah, 2015-08-07:
No, that would be you updating the key and hiding the problem by changing the install folder. We'll still have the intermittent problem with others.

bsongis, 2015-08-07:
It would be interesting to dump the registry before and after 2.1.1 uninstall, and then check the difference.

rguptaca, 2015-08-07:
Did I not post the two Registry files?? I'm sure I did. In two different comments because there was no attache file option for me.

kilrah, 2015-08-07:
Yes I deleted that as it took pages and was unreadable, we only need one or 2 keys from that. But I'll look at it, I've actually seen it on one of my machines myself..

jmrware, 2015-08-08:
I see that you guys already know about this one. I wrote about it on the RCG 2.1 thread here:

http://www.rcgroups.com/forums/showpost.php?p=32364733&postcount=489

Bottom line is that if you install 2.1.0 (or 2.1.1) on top of 2.0, this first install works ok, but the pump is primed for the next upgrade clobbering the 2.0 install. (The "default" and "Start Menu" paths/keys in the HKCU-Software-OpenTX-Companion 2.1 registry folder erroneously point to the 2.0 folders.)

kilrah, 2015-08-08:
OK, it is what I thought it was.
If 2.1 has never been installed or run but a version of 2.0 is present:
- The installer looks for "HKCU\Software\OpenTX\Companion 2.1" "", which does not exist so it uses the default install directory ($PROGRAMFILES\OpenTX\Companion 2.1) and start menu folder.
- At the end of the install it sets this reg value plus the start menu one to the actual directory companion got installed in for next time (user might have changed it). All good up to here. 

[image]

Now when companion 2.1 is run it sees it has no settings yet, and imports settings from any older version found which is fine, BUT the bug is that the current copy method copies ALL settings blindly and including the install dir and start menu folders from the old version, overwriting the correct ones that were set by the installer. So if a version of 2.0 is installed the install dir will become that of 2.0, which the installer will use next time it's run.

So the solution is simple, the loops that copy settings in appdata.cpp:614 and appdata.cpp:630 must be updated to exclude those 2 values (the default one and the "start Menu Folder" one).

The user will still have to be careful and check/fix the install folder on the 2.1.2 update.

commit 08756af9 (2015-08-08): Don't import install dir settings from old versions (fix #2645)
commit 5d10ca62 (2017-02-20): [Companion] Exclude Windows installation-specific keys when importing settings from previous versions (truly fixes #2645).
commit d8b4f859 (2017-02-28): [Companion] Smarter app settings import from old versions... (#4485)

* [Companion] Smarter app settings import from old versions; Do not import flash tool locations on Windows which are already included with Companion (closes #4330).

* [Companion] Exclude Windows installation-specific keys when importing settings from previous versions (truly fixes #2645).
closed 2015-08-08 by kilrah