opentx/opentx #2646: LUA:  A function to query the units defined in the sensor

fraternl, 2015-08-06:
In 2.1 you introduced the function to query the radiosettings of the preffered unit settings (getGeneralSettings().imperial ).
This is important so we now know the user's preference (metric/imperial)

But because now the sensors have their own setting for the units as well it's important to know what unit it is set to....

Is there a function to query that value?

I also have a question about that unit setting.
Is it something that has to match the output of the sensor and is it mandatory that it matches the sensor or is a conversion made if it is manually changed?

Because the units can be changed to any unit (even degrees Farenheit for a vario) I would think it's only a helper function for the system to know which units should be used. Changing it to other values would just give erratic results. Am I correct here?

fraternl, 2015-08-06:
http://www.rcgroups.com/forums/showthread.php?t=2180477&page=90#post32349431

kilrah, 2015-08-06:
If the unit is changed to a compatible one a conversion is made, we know from somewhere else what we receive.

projectkk2glider, 2015-08-06:
First of all I think the `g_eeGeneral.imperial` is obsolete for ARM platforms with the new telemetry system and should be removed.

In the new telemetry system each telemetry value can have its own unit so something needs to be done in the Lua API to get that information across.

kilrah, 2015-08-06:
The general setting was kept to choose the default unit for newly created sensors. Some types may be missing that step though.

projectkk2glider, 2015-08-06:
@kilrah  I am not seeing any (relevant) usage of that setting in current code for ARM platforms.

fraternl, 2015-08-06:
This also means that the 2nd parameter for PlayNumber( Alt, 9 ) should be ignored... or not?

projectkk2glider, 2015-08-06:
>This also means that the 2nd parameter for PlayNumber( Alt, 9 ) should be ignored... or not?

No, you are playing some number with some unit. The radio can not know that you are playing altitude in this case. We would have to have a separate function to play a particular telemetry value, then the unit could be automatically played along.

fraternl, 2015-08-06:
Will this get a 2.1.1 milestone?
A script in development which targets both 2.0.x and 2.1.x would otherwise need special handling for the current 2.1.0
My preference would be to drop support of 2.1.0 and make it only compatible with 2.0.x and 2.1.1 and higer.

kilrah, 2015-08-06:
> @kilrah I am not seeing any (relevant) usage of that setting in current code for ARM platforms.

frsky_sport.cpp:285. 
Indeed we need to expand it for m/s, °C etc

> Will this get a 2.1.1 milestone?

Most likely not, 2.1.1 has to be out tonight.

projectkk2glider, 2015-08-06:
>frsky_sport.cpp:285.

Oh, sorry, missed that one. 

But then the `frskyDSetDefault()` is missing this part of the code and some other parts too. 

```c
void frskySportSetDefault(int index, uint16_t id, uint8_t instance)
{
  TelemetrySensor & telemetrySensor = g_model.telemetrySensors[index];

  telemetrySensor.id = id;
  telemetrySensor.instance = instance;

  const FrSkySportSensor * sensor = getFrSkySportSensor(id);
  if (sensor) {
    TelemetryUnit unit = sensor->unit;
    uint8_t prec = min<uint8_t>(2, sensor->prec);
    telemetrySensor.init(sensor->name, unit, prec);
    if (id == RSSI_ID) {
      telemetrySensor.logs = true;
    }
    else if (id >= ADC1_ID && id <= BATT_ID) {
      telemetrySensor.custom.ratio = 132;
      telemetrySensor.filter = 1;
    }
    else if (id >= CURR_FIRST_ID && id <= CURR_LAST_ID) {
      telemetrySensor.onlyPositive = 1;
    }
    // TODO set telemetrySensor.onlyPositive for other sensors that need it
    else if (unit == UNIT_RPMS) {
      telemetrySensor.custom.ratio = 1;
      telemetrySensor.custom.offset = 1;
    }
    else if (unit == UNIT_METERS) {
      if (IS_IMPERIAL_ENABLE()) {
        telemetrySensor.unit = UNIT_FEET;
      }
    }
  }
  else {
    telemetrySensor.init(id);
  }

  eeDirty(EE_MODEL);
}
```
```c
void frskyDSetDefault(int index, uint16_t id)
{
  TelemetrySensor & telemetrySensor = g_model.telemetrySensors[index];

  telemetrySensor.id = id;
  telemetrySensor.instance = 0;

  const FrSkyDSensor * sensor = getFrSkyDSensor(id);
  if (sensor) {
    TelemetryUnit unit = sensor->unit;
    uint8_t prec = min<uint8_t>(2, sensor->prec);
    telemetrySensor.init(sensor->name, unit, prec);
    if (id >= D_A1_ID && id <= D_A2_ID) {
      telemetrySensor.custom.ratio = 132;
      telemetrySensor.filter = 1;
    }
    else if (id == D_RSSI_ID) {
      telemetrySensor.filter = 1;
    }
    else if (unit == UNIT_RPMS) {
      telemetrySensor.custom.ratio = 1;
      telemetrySensor.custom.offset = 1;
    }
  }
  else {
    telemetrySensor.init(id);
  }

  eeDirty(EE_MODEL);
}
```
Should this be fixed?

kilrah, 2015-08-06:
Yep sure. D telemetry was a bit left behind, like the small screen platforms that are getting fixed these days. I don't really have much D-type stuff left to test with...

projectkk2glider, 2015-08-06:
Ok then, I am still on D stuff, so I will fix this in #2652.

adrionics, 2015-09-16:
I would like to add another voice to request this function so we can use the sensor values from LUA and be able to handle units correctly.

From reading this thread, I am not sure if it has been agreed that this function is needed?

closed 2016-11-27 by fraternl