opentx/opentx #2647: Feature - Range Check Switch Assignment

rguptaca, 2015-08-06:
Since it is recommended that flyers do a range check before flying it would be useful to be able to access it quickly via a switch on the Tx.

fraternl, 2015-08-06:
Good one....
But I wouldn't recommend it to get linked to a physical switch.
A logical switch that creates a virtual latch is recommended
http://rc-soar.com/opentx/setups/vlatch/vlatch.htm

bsongis, 2015-08-06:
If I am not mistaken, I already implemented it

fraternl, 2015-08-06:
I can only see "instant trim"

kilrah, 2015-08-06:
It is already implemented, but you need to compile the firmware yourself with DANGEROUS_MODULE_FUNCTIONS=YES for obvious reasons.

closed 2015-08-06 by kilrah

rguptaca, 2015-08-06:
I posted after looking at every option in special functions and global functions using companion and Tx - N407.

rguptaca, 2015-08-06:
There are numerous ways for us to screw up. Why would range check be any different.

fraternl, 2015-08-06:
Totally agree with rguptaca
Can't that be set in Companion????

rguptaca, 2015-08-06:
Come on guys. I know you work hard. I know getting your worked picked apart can be annoying. We are just trying to help. Closing an issue with no further discussion is a bit dismissive of us trying to help.

kilrah, 2015-08-06:
It was discussed several times already before you got there.

We know that the way we use to enable it disable a mix works and can rely on it.

We however don't know how a 3rd party module reacts to something it was not meant for like switching in/out of range check mode in flight should you hit the switch by mistake.

rguptaca, 2015-08-07:
Thanks.
Decision has been made:)
I will have to move on... I don't do range check's anyway after using a Rx for the first time; too cumbersome. 

Good work guys!

fraternl, 2016-02-18:
It is quite frustrating to ask for a feature and then find out "it's been done already, but you can't have it"
Not entirely true, as I could compile my own firmware, but close to it.

I don't really understand why it is withheld at all. This special function has to be explicitly entered in the SF's. How can this be dangerous?

I want to encourage myself in doing a range check, because I want to automatically enable this for FM5
FM5 is my calibration mode. In this mode it's impossible to fly the model.
In practice I hardly ever do a range check and rely on RSSI for this. 

> We however don't know how a 3rd party module reacts to something it was not meant for like switching in/out of range check mode in flight should you hit the switch by mistake.

I will not put this on a physical switch.
It will be triggered by a logical switch (in my case FM5) 
If it would enter FM5 during flight I might even be better off when it would become out of range (failsafe).

Could you please better explain why these are withheld?
Because you seem to be (overly) concerned about this I proposed this one 6 months ago
https://github.com/opentx/opentx/issues/2648
but nothing is done with it.

kilrah, 2016-02-18:
This issue is closed because it is a duplicate of the one you just linked to... which hasn't been closed, and even has a milestone set which means it's not rejected and is just lingering in the mountain of "to do one day" things.