opentx/opentx #2648: Another checkbox in Companion to enable dangerous functions

fraternl, 2015-08-06:
It seems some "dangerous functions" are withheld (with best intentions of course) in the regular firmware.
https://github.com/opentx/opentx/issues/2647

I would like to be able to select these by enabling a checkbox in Companion, just like "nogvar", "lua"....

rguptaca, 2015-08-06:
100% agree.
What is dangerous?
Who is deciding?
Is it dangerous to be able to assign switched to THR AIL ELE RUD? What if I use the wring switch while flying. There goes my plane.
It is one thing to help us be safe it is quite another deciding for us.
The developers are doing awesome work.

bsongis, 2015-08-06:
Yes this needs to be discussed, but later, I will focus on 2.1.1 release tonight

rguptaca, 2015-08-06:
Thank you:)

Rav

Have a wonderful day!

Ravi Gupta, PMP
Founder of Hike For Happiness
A Mental Health Fundraiser
http://hikeforhappiness.ca/
(519) 697-0796

fraternl, 2015-08-10:
If I were to have a say in this I would vote for addition of this function to all versions of the platforms that permit this function. Not even this checkbox in Companion.

I'm planning to use this function on my DLG whilst it is in Flightmode 5
This flightmode 5 is a calibration mode. This mode is already a "dangerous flying mode" 

Almost no-one has a development environment to compile the firmware themselves.

Waldo-P-Barnstormer, 2015-08-12:
If you want a quick, and safe way to get to Range check, what about changing the order on Pg 2 - Model Setup. (Taranis)
Put the  "Receiver		[Bind]    [Range]"   line ahead of Model Name. and have [Range] first on the line.  "Receiver		[Range]    [Bind] "    then  [Range]  check will become the default on the page.
A push of the Menu key, Page key, and you're there.
If you've selected a different model, simply hit Page key, Enter key and you'd be in Range check mode.  Long press on Exit when you're done.

I am wondering if enabling Logical Switches for Global Functions would be possible.
I currently use 8 LS for each aircraft, 1 to start logs, 2 for Play Value functions, and 5 for Play Track functions.
Maybe enabling  a "Dangerous Functions"  in Build Options that would allow this could be an idea.
I see in OpenTX 2.2.X an open issue for "Adding global logical switch #2409"  that will likely allow logical switches for Global Functions but 2.2.X is a fair ways off in the future.

rguptaca, 2016-05-12:
Is there an ETA on this? No rush... just asking :)

closed 2016-11-27 by fraternl