opentx/opentx #2657: Downloading Dutch firmware gives "Temporary error" on server (Jenkins?)

fraternl, 2015-08-06:
Downloading Dutch firmware gives "Temporary error" on server (Jenkins?)

This may be even my fault..
Kilrah recently added an "improved" nl.h.txt 
I now wanted to check it on the radio to see if I can find things before the release....

Would like to have it properly working on 2.1.1 and will work on it....

I hope you can give me a hint where it might have gone wrong...

bsongis, 2015-08-06:
I just see that the compilation server failure notifications go to my spam folder!

bsongis, 2015-08-06:
translations/nl.h:935:45: warning: missing terminating " character [enabled by default]
translations/nl.h:1046:32: warning: missing terminating " character [enabled by default]

fraternl, 2015-08-06:
#define TR_TRAINER             Trainer Poort"
#define TR_COPY_STICKS_TO_OFS  "Kopieer Sticks naar Subtrim"  
#define TR_COPY_TRIMS_TO_OFS   "Kopieer Trim naar Subtrim"
#define TR_EEBACKUP            "\004[ENTER Lang] voor een EEPROM backup"
#define TR_WAV_VOLUME          "Wave-Volume"
#define TR_CELLINDEX           "Cellenindex"
#define TR_IGNOREIDS           INDENT "Ignore instance"

made these changes...
ah.... 2 missing double quotes...  will check the other one and make it available

fraternl, 2015-08-06:
Should be corrected now in http://hmvc.eu/nl.h.txt

kilrah, 2015-08-06:
We REALLY encourage translators to have a build environment to catch this kind of errors...

closed 2015-08-06 by bsongis

fraternl, 2015-08-06:
I guess I now have to wait until tomorrow....
I was hoping to check things to see if it needed more corrections.

I do understand I'm an odd one out. 
Sorry for that

commit fce1cc68 (2015-08-06): Fixes #2657