opentx/opentx #2661: MAC OS X Companion 2.1.1

rguptaca, 2015-08-07:
OS X Ver 10.10.1
Mac Mini

Companion 2.1.1 public release

MAC OS behaviour 
Minimise window -> do something else -> Click OpenTX icon on dock -> app window does not show up 
Usually the app window shows up.

I had to right click on the OPEN TX dock icon and select Show All Windows.

rguptaca, 2015-08-07:
ls: /Users/ravigupta/Downloads/OpenTX*.csv: No such file or directory
===========================
OpenTX Voice Sound Pack Generator - Mac OSX - v1.2 - 2015/07/15 - Written By: Austin St. Aubin (AustinSaintAubin@gmail.com) 
   ∟ Description: Description: Creates phrases with macs osx's built in say command based on cvs file.
~ - - - - - - - - - - - - ~
Destination Directory: /Users/ravigupta/Downloads/SOUNDS
Source File: 
~ - - - - - - - - - - - - ~
Hello, my name is Samantha, I would create your sound pack but I can't find the source file needed to create it. Please check your configuration in the script, and the title of your source file.

Suggestion 
- look for *OpenTx*.csv
- implement script with source file as optional
 sh opentx_voice_and_sound_pack_generator_script_mac-osx.sh Copy\ of\ OpenTX\ 2.x\ -\ Voice\ and\ Sound\ Pack\ Generator\ -\ Definitions\ by\ AustinSaintAubin\ -\ MAIN.csv 

Yes - I know renaming the file will fix this.... just a suggestion.

bsongis, 2015-08-07:
This tool is not part of OpenTX!

kilrah, 2015-08-07:
And it's completely off topic for the issue...

rguptaca, 2015-08-07:
Sorry about that.

rguptaca, 2015-08-07:
Dont know why I put it here. How do I delete it?

projectkk2glider, 2015-08-07:
Unrelated posts removed.

closed 2021-04-22 by kilrah