opentx/opentx #2662: "Bricked" radio after flashing

fraternl, 2015-08-07:
This is the 2nd time I "bricked" my radio.
This time I had the Dutch firmware on it, but didn't like it  ;-)
Then I flashed the normal English version on it..

After I disconnected the USB I was able to chose "Exit" and I got a blank screen.
Then I turned it off....  nothing.... Turned it on....   (My Taranis has a "power LED" and it came on), still nothing...   I pulled the battery, waited a while...  put it back in, turned it on...   nothing...

Then I pulled the battery, flashed it and after I put the battery back in and turned it on everything was fine again.

This was all with the current 2.1.1 release
The bootloader was a recent nightly... I believe something with 409
AFAIK nothing changed in the bootloader for a while (since 2.0.17)
It's now on 2.1.1 because of the latest flash

kilrah, 2015-08-07:
What computer OS?

fraternl, 2015-08-07:
Windows XP

kilrah, 2015-08-07:
You're not clear whether you flashed with dfu or from within the bootloader, seems a bit of both as your bootloader is updated but we don't know what you used when.

fraternl, 2015-08-07:
I thought it was clear by its context..

I flashed with 2.1.1 companion whilst the radio was in bootloader (2.1.1 nightly)...

After I bricked it I used companion again, but because the battery was disconnected and the radio therefore off it (companion 2.1.1) flashed it with DFU.

The radio started working again afterwards.

bsongis, 2015-08-07:
Taranis X9E ?

fraternl, 2015-08-07:
No... no...  an X9D
Just wanted to mention it....
You worked on it before and it didn't happen since....
Just wanted to give you a heads up....   
I know how to solve it, luckily

bsongis, 2015-08-07:
Is it reproductible? In case it was a transfer error ...

projectkk2glider, 2015-08-07:
I think the reason is the same as in #1997. The current method of flashing via fake drive is not reliable, the OS is free to send the file chunks in any order and free to rearrange them on the block device, which his screwing up the flashing process. We need alternative ASAP, the #2008 is most relevant issue that I can find at the moment.

bsongis, 2015-08-07:
Yes

closed 2015-08-07 by bsongis