opentx/opentx #2665: Sound files are "unintelligible" on 9xr-PRO since 2.1

pyrou, 2015-08-07:
I've tested openTX 2.1 and 2.1.1 with almost every english sound packs available on voices-21.open-tx.org/opentx-9xrpro/en/ (even those uploaded today 07-Aug-2015 at 00:34). I also tried the old one on voice-20.

In any test case, sound files looks like being played at a wrong sampling rate. 
here is a "demo" of what I'm hearing : https://soundcloud.com/pyrou/opentx-sound-bug-on-9xrpo/s-5SPqM

Back to openTX 2.0 : everything "playing" normally

bsongis, 2015-08-07:
Yes I reproduce it. I thought I didn't change anything. I will find the wrong commit.

bsongis, 2015-08-07:
It comes from #2288. I moved the audio buffers out of the AudioQueue class.
The thing is if I put them again in the class, it works, if they are outside the class it is broken, the sound is really bad, it seems a bad frequency. 

I checked the initialization in case the constructor was not called
I tried to make some more variables volatile

For the moment no luck, I don't want to put the buffers again in the class. Any idea welcome!

bsongis, 2015-08-07:
I also checked the alignment

bsongis, 2015-08-07:
Now why this fixes the issue, I don't understand
@MikeBland @projectkk2glider an idea?

bsongis, 2015-08-07:
I just had a look in mbtx/ersky9x and I see this code:

uint32_t x = DACC->DACC_TNPR = CONVERT_PTR_UINT(nextBuffer->data);
DACC->DACC_TNCR = nextBuffer->size/2;
*DACC->DACC_TNPR = x;		// Goes wrong without this!!!*

With this hack, it's better, but still not perfect, that's really weird, no?

projectkk2glider, 2015-08-07:
Check the memory map before and after your fix.

projectkk2glider, 2015-08-07:
If I build before your latest fix it get badly aligned buffers:
```
.bss.audioBuffers
                0x20000836      0x78c allsrc.o
                0x20000836                audioBuffers
```

They should be on 4 byte boundary. 0x20000836 /4 = 0x800020D,8

projectkk2glider, 2015-08-07:
Ignore my previous comment, I have found no document that states that buffer should be aligned.

bsongis, 2015-08-07:
Yes I checked the alignment, it's exactly the same result when it's aligned on 32bits

bsongis, 2015-08-07:
I think I just found a better fix. And yes it should be aligned. The interrupt seems called more than I thought...

closed 2015-08-07 by bsongis

pyrou, 2015-08-10:
Works for me with e136905 
Many thanks for your reactivity !