opentx/opentx #2670: Feature request: Highlight unreferenced logical switches

fraternl, 2015-08-08:
If a logical switch is unreferenced by either a special function or another logical switch this may indicate an orphaned switch or user error..
When fiddling around with different kind of setups and new ways to achieve something it is of great help if these are highlighted.

In Companion this could be in red and on the radio underscored

projectkk2glider, 2015-08-08:
I am not in favor of this kind of check on the radio, too much additional code.

But it could be done in the Companion, I envision a kind of *model checker* function that could be executed from the menu and then a report created that would point out any obvious problems with the setup like unreferenced switches, etc...

fraternl, 2015-08-08:
On Companion it could be done in realtime, no?

projectkk2glider, 2015-08-08:
Yes it could be done that way, but I prefer to gather all errors (if we do a more complete check) in one place.

kilrah, 2015-08-08:
>On Companion it could be done in realtime, no?

It would become way too annoying/intrusive, the moment you set up an LS it would come up in red because it's obviously not used yet...

fraternl, 2015-08-08:
Who's afraid of red yellow and blue?

closed 2016-11-27 by fraternl