opentx/opentx #2671: Taranis-E upper display voltage meter too fast

janekx, 2015-08-08:
Sometimes happend when battery is 9.65V the upper display show 9.6 and 9.7V too fast that finally it shows both together. In lower display it shows well stable voltage. It needs to make some filter.

projectkk2glider, 2015-08-09:
I can not reproduce. The battery voltage is already filtered and should not change so rapidly. 

What is possible is a bug in a top LCD driver, some other part of drawing functions could be messing up the voltage display. Could you make a video of the problem.

Spainman, 2015-08-09:
A somehow related observation: The bar display for the battery on the top display is not in sync with the LCD although they have the same number of bars. The top display tends to show one bar less unless the battery is completely full.

janekx, 2015-08-10:
Will try to make video,  hapend to me two times in 8h battery life it is some rare situation bug.

projectkk2glider, 2015-08-10:
I just witnessed this the voltage was changing rapidly and the battery meter symbol also. I was in analog values screen and as soon as I pressed EXIT key the blinking stopped and voltage was table at 7.9V. 

I also notice a bug in the battery bars display on top lcd. I the battery range is completely outside the current battery voltage the indication is sometimes wrong - it shows full battery.

janekx, 2015-08-11:
Good so I can leave hunting this and going to focus on other one :) 
Battery bars issues I not noticed.

projectkk2glider, 2015-08-11:
I can explain (well at least my incident). This happens when one is on ANALOG INPUTS radio screen where the battery voltage can be calibrated. When radio is on this screen the voltage is calculated and updated much faster than normally and that can cause rapid changes of the numbers. It is perfectly normal. 

@janekx did you also experience this issue on this screen?

The other problem with the battery gauge still remains to be fixed.

Spainman, 2015-08-11:
Here is a picture showing my observation regarding different numbers of bars on the top and main display (3 vs 4). The firmware is stock as it came from frsky. Settings for the indicator range was 9.2 - 11V, battery alarm 9.4V. I already sold the x9e, so I can't provide further information. Hope it helps anyway.
[image]

janekx, 2015-08-12:
Sorry I am not sure if was on analog screen at this moment I thinks that no because I compare it with upper voltage reading on main model page, but not 100% sure, in any case I will try to reproduce it.

closed 2015-08-28 by kilrah