opentx/opentx #2672: Taranis-E sugestions for upper display

janekx, 2015-08-08:
I have these sugestions for upper display. 

Operation time usable as second timer and both timers make possible to blink in case of overtime or time event.

Battery metter and voltage to fill with data from telemetry so can show model battery. In case of battery low can the battery meter blinks. 

In case of shows radio battery blinks the red segment of the meter when battery going low.

RSSI numbers will be RSSI it is good to see.

RSSI bars can be switchable to vario rising and faling when two yelows are 0 and green rising and red falling. 

Any other sugestions welcomed.

hrenz, 2015-08-08:
Hy,
upper color display is a botch
nothing is readable  outside at daylight!

If Horus will be the same then forget it!
Futaba and Graupner colordisplay same botch, nothing is readable
Look at Jeti and Multiplex both have large B/W LCD
all is fine readable outside at a sunny daylight

janekx, 2015-08-08:
This is issue to what possible make with display not the HW issues !!!

kilrah, 2015-08-08:
Deleted... but yes there's a point. I haven't seen production hardware myself but most reports I've seen indicate the top LCD is unreadable, so there might be little sense in spending time customizing it. 

What do you think of it?

janekx, 2015-08-08:
Yes, please look at my youtube channel dedicated to Taranis-E https://www.youtube.com/playlist?list=PL8pjlpRO1RIJEKNXUGdNdVd6GdKa5psMe
here is video from sunny day looking at upper display https://youtu.be/cEckKU88YBc?list=PL8pjlpRO1RIJEKNXUGdNdVd6GdKa5psMe&t=106
In real it is little better visible then on camera but yes it is poor.

hrenz, 2015-08-08:
Hy, 
outside at the sun you see absolut NOTHING at the upper Display.
Lower B/W LCD is fine at the sun.

Did nobody from FrSky test it outside at the sun?

kilrah, 2015-08-08:
>Is it not allowed to say this here?

No because it has absolutely nothing to do with OpenTX. If you want to complain do so at FrSky, they're the ones who decided to use this display (against our advice, by the way) and they're the ones who can do anything about it.

BTW it's nothing like a real color LCD, it's a monochrome segment display with a weak white backlight and a color filter. So not representative of anything that could be used on Horus.

hrenz, 2015-08-08:
Hy kilrah,
sorry, correct, nothing to do with openTx

janekx, 2015-08-08:
Please leave my suggestion clear from issues regarding to FrSky.  This is suggestions how to use upper display...  My videos shows that there is poor visibility as well but my display is covered with antreflex tablet shield and no the transport foil.

bsongis, 2015-08-08:
I removed some posts in this thread. The discussion is about what we can do with the top Lcd we have. For the moment I did something quick to be on time. It can be improved and all ideas are welcome.

closed 2020-07-08 by stale[bot]