opentx/opentx #2678: Companion 2.1.1 does not open with OSX 10.11 (El Capitan - Beta)

tonimerlot, 2015-08-09:
Companion 2.1.1 does not open with OSX 10.11 (El Capitan - Beta)

------------------------------------------------------------------------------------------------
Process:               companion [1562]
Path:                  /Users/USER/Downloads/*/companion-2.1.1.app/Contents/MacOS/companion
Identifier:            companion
Version:               ???
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           companion [1562]
User ID:               501

Date/Time:             2015-08-09 12:19:52.257 +0200
OS Version:            Mac OS X 10.11 (15A243d)
Report Version:        11
Anonymous UUID:        1DF4459C-D5C4-DC65-A9D6-6551E41EA88C

Time Awake Since Boot: 1800 seconds

Crashed Thread:        0

Exception Type:        EXC_BREAKPOINT (SIGTRAP)
Exception Codes:       0x0000000000000002, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
dyld: launch, loading dependent libraries

Dyld Error Message:
  Library not loaded: @executable_path/../Frameworks/SDL.framework/Versions/A/SDL
  Referenced from: /Users/USER/Downloads/*/companion-2.1.1.app/Contents/MacOS/companion
  Reason: image not found

Binary Images:
    0x7fff63056000 -     0x7fff6308cbf7  dyld (360.11) <8D611E1A-C060-3E57-82F1-E952FB6908C8> /usr/lib/dyld
    0x7fff8c4cb000 -     0x7fff8c4ccff3  libSystem.B.dylib (1225) <B23C2B8E-5093-327F-A683-20F8421E3608> /usr/lib/libSystem.B.dylib

Model: iMac14,2, BootROM IM142.0118.B11, 4 processors, Intel Core i5, 3,2 GHz, 8 GB, SMC 2.15f7
Graphics: NVIDIA GeForce GT 755M, NVIDIA GeForce GT 755M, PCIe, 1024 MB
Memory Module: BANK 0/DIMM0, 4 GB, DDR3, 1600 MHz, 0x02FE, 0x45424A3430554738454655352D474E4C2D46
Memory Module: BANK 1/DIMM0, 4 GB, DDR3, 1600 MHz, 0x02FE, 0x45424A3430554738454655352D474E4C2D46
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x111), Broadcom BCM43xx 1.0 (7.21.91.0.3a3)
Bluetooth: Version 4.4.0d86 16188, 3 services, 27 devices, 1 incoming serial ports
Network Service: Ethernet, Ethernet, en0
PCI Card: pci1b21,612, AHCI Controller, Thunderbolt@195,0,0
Serial ATA Device: APPLE HDD WD10EZES-40UFAA0, 1 TB
Serial ATA Device: ST500LM012 HN-M500MBB, 500,11 GB
USB Device: USB 3.0 Bus
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: FaceTime HD Camera (Built-in)
USB Device: ADC
Thunderbolt Bus: iMac, Apple Inc., 23.10
Thunderbolt Device: HD-PATU3, BUFFALO INC., 1, 5.1

------------------------------------------------------------------------------------------------

kilrah, 2015-08-09:
Leaving this one open because 3 per day starts to be a bit annoying. Hopefully people will see it before posting.

bsongis, 2015-08-10:
Is it possible that someone tests my new .dmg which I just uploaded here:
http://downloads-21.open-tx.org/companion/companion-macosx-2.1.1-new.dmg
Here are the tests to do:
- check the installation process runs fine
- test the simulators
- check that the sound is OK

floaaaaargh, 2015-08-10:
Hi, thanks for uploading. 
• the installation runs flawless
• the simulation of the taranis works (I just started it an flicked a few switches assined to SFs)
• the simulator also plays sound

As just beeing a user and not a dev, I cant see any issues so far.

kilrah, 2015-08-10:
I'll be able to test in an hour or so

kilrah, 2015-08-10:
All good for me!

commit 2d9c1308 (2015-08-10): Merge pull request #2687 from opentx/bsongis/osx

Mac libraries now loaded correctly (Fixes #2678)
commit a569c673 (2015-08-10): Fixes #2678
closed 2015-08-10 by kilrah