opentx/opentx #2679: External module power?

lvale, 2015-08-09:
Radio is Taranis X9D (B). External module is an OrangeRX DSMx (this one http://orangerx.com/2012/10/31/dsmxdsm2-2-4ghz-transmitter-module-jrturnigy-compatible/ )

Radio version is 2.11

Created a new model from scratch on the radio, and on the model settings switched off the internal module and enabled the external module as PPM.

As soon as I change on the external line from OFF to PPM the module switches ON and I can see the LED on the module flashing.

The issue comes when we turn the radio off.

When I switch the radio ON (on the respective model) the module does not power up. I can switch to other models that use the internal one and those work fine, but when I switch back to the model that uses the external module it won't power the external module.

The only workaround I've found is to go to the menu, switch off the external module and back on to PPM. Then the external module "behaves" as expected.

I didn't test this on previous versions of OpenTX, so I'm not sure if this a 2.11 behavior only, but I don't think this is a "good" behavior?

kilrah, 2015-08-09:
Just tried on 2.1.1 now with same settings and it works as intended i.e. module power comes on on power up after warnings are clear.

If you've got a multimeter please check whether the power supply is there on the pins when doing the same. I'd suspect the module does strange things.

Failing that please supply all details about your setup (which firmware options, and if possible the EEPROM .bin backup file from your radio).

lvale, 2015-08-09:
ok
Just tested the VCC on the pins and it has power, but the module still won't transmit, not even to bind like on this video
https://www.youtube.com/watch?v=YIHlV7mibuQ

If I need to bind I go to the menu and change the external module to off and back to PPM again while pressing the bind button on the module.

Unfortunately I don't have a way to check if there is a PPM signal....

the files are here:

https://www.dropbox.com/s/5cn1e8czuq0xcng/211%20eeprom%20and%20model.zip?dl=0

lvale, 2015-08-11:
Just an additional note:

IF the PPM is set to CH1 to 8 the module powers ok. BUT if the channel is set to CH1-16 it does not.

So it only occurs if the External Module is set to PPM with CH1-16.

Strange :)

projectkk2glider, 2015-08-11:
Are you sure the module you have supports 16 PPM channels? According to http://www.hobbyking.com/hobbyking/store/uploads/1018410990X893272X50.pdf it does not:
```
PIN1- PPM signal, TTL 5V, 6-12 channels
```

lvale, 2015-08-12:
The receiver I'm using has SBus out and indeed all 16 channels are received when I do that "off/on" dance on the menu.
The specs might also apply to this module, but it is a different (previous) version.

kilrah, 2015-08-28:
Closing as I can't see how this can be an OpenTX problem.

gereic, 2019-11-14:
Hi, I want to reopen this issue.
I have the same problem on the same module.
The problem is, that the ppm-pin has a high-level before the battery-power is on the module.
Is it possible to set ppm-signal after module is powered up (may with delay of 100ms) ?

bsongis, 2019-11-14:
are you sure? how do you know this?

gereic, 2019-11-14:
I was trying it.
When I plug in power on the module and then the ppm, it is working. When all is connected and powered on, not working.
Also, when you switch the external module off, ppm Pin has 3.3V.

bsongis, 2019-11-14:
Are you in 2.3 ?

gereic, 2019-11-14:
2.3.1

bsongis, 2019-11-14:
are you able to compile yourself? I would give you a patch to try

gereic, 2019-11-14:
Is there a wiki what I need to compile?

bsongis, 2019-11-14:
yes there is

gereic, 2019-11-14:
Ok, then it should be no Problem.
You can give me the patch and I will try it.

bsongis, 2019-11-21:
When enabling the PPM module with menus:
[image]

When starting the radio:
[image]

gereic, 2019-11-21:
Yes, that's the problem.
As you can see the PPM-Pin has High-Signal even when there is no Power.
So the solution would be to set the PPM-Signal to low, when there is no power-supply.
But this is special for the orange-tx-modules.